ihmcrobotics/ihmc-open-robotics-software #16: IHMC Log Files should log stdout/stderr

dljsjr, 2016-03-02:
IHMCLogger should log stdout/stderr so we can get at output from runs when robots are not being managed using IHMC Mission Control (currently, the only robot managed by IHMC Mission Control is Atlas).

There is an internal JIRA ticket on the IHMC JIRA for this, but as JIRA isn't public yet and we don't have a nice way of bridging our JIRA with GitHub Issues I wanted to make this issue as well.

IHMC JIRA key is ROSI-381.

dljsjr, 2016-03-02:
Test comment, please ignore.

dljsjr, 2018-08-01:
I'd still like to see this. I'm making a maintenance/feature improvement pass on the logger soon.

calvertdw, 2018-10-18:
The new log-tools library might help: https://github.com/ihmcrobotics/log-tools

closed 2023-12-30 by calvertdw