ihmcrobotics/ihmc-open-robotics-software #17: Minimal pose changes lead to qdd_d close to 0, freakout

iamwolf, 2016-03-03:
This is a follow up on the Valkyrie controller freakout issue reported last night by @mauricefallon and @jlack1987. I was able to reliably recreate a similar situation in simulation.

Observation: if sending a second trajectory that is only minimally different from the first close to joint limits, the qdd_d values are all 0 (or very close to it), and the [robot freaks out/simulator stops].

The console output is:
```
Loading robot model from: 'models/val_description/sdf/valkyrie_sim.sdf'
[INFO] (NetworkParameters.java:34): Loading network parameters from /home/wxm/.ihmc/IHMCNetworkParameters.ini
[WARN] (JointStateUpdater.java:90): Could not find the given pelvis and/or chest IMUs: pelvisIMU = pelvis_pelvisRearImu, chestIMU = torso_leftTorsoImu
[WARN] (JointStateUpdater.java:98): Pelvis IMU is null.
[WARN] (DRCSimulationFactory.java:276): Initializing Estimator to Actual!
[INFO] (DRCNetworkProcessor.java:290): Connecting to controller using intra process communication
[INFO] (OffscreenBufferVideoServer.java:39): Starting video stream
[INFO] (JMERenderer.java:859): GPULidar scene updated. Took 5.4E-4 s
[INFO] (JMERenderer.java:766): A PBOAwtPanel showed on screen.
[INFO] (JMERenderer.java:859): GPULidar scene updated. Took 1.5E-5 s
IHMC ROS API node successfully started.
[WARN] (HandControlModule.java:914): Waypoint 18 is out of the joint position limits, cancelling action.
```

The git revision is 3e86994d26745696745a706e74c4e5067f23a703. The git blame shows the last lines that output any message to be from the 24th of January, Valkyrie Operator Computer - cf. [here](https://github.com/ihmcrobotics/ihmc-open-robotics-software/commit/7501170a445ab527debca0a227a54cd891437900)).

**Here's a [minimal rosbag](https://drive.google.com/file/d/0B_k3bV-5JKanOGRUaFhZMnNHRW8/view?usp=sharing) to recreate it - just start the ValkyrieROSAPISimulator or OpenHumanoidsSimulator, start playing the rosbag and watch when the LIDAR stops spinning/the HandControlModule message appears.**

**Here's the corresponding [video](https://www.youtube.com/watch?v=t0Tou6lWscA); watch how the SCS threads die out** [we've reproduced this on a laptop, our workstations have more RAM and CPU, so that wasn't the issue]

Can I provide any other logs that would be helpful?

cc: @dljsjr @SylvainBertrand

iamwolf, 2016-03-03:
Another observation: Commenting out the ``trajectoryGenerator.clear()`` [here](https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/develop/CommonWalkingControlModules/src/us/ihmc/commonWalkingControlModules/highLevelHumanoidControl/manipulation/individual/HandControlModule.java#L915) and simply returning does not cause the threads to crash. @SylvainBertrand, would it be safe to comment out the clear and simply return?

SylvainBertrand, 2016-03-03:
Yo! Alright I'm on it, I'll let you know when I got the fix in.

iamwolf, 2016-03-03:
Thanks! Also keen to know what the implications of a simplistic fix such as #18 would be (just verified that this would still work with the new changes Doug pushed).

SylvainBertrand, 2016-03-03:
I won't use your fix. Instead, I'll do what I was actually expecting the code to do: 1- check the data (positions within limits and increasing waypoint time when going through the waypoints in order), 2- only if the data is acceptable start setting up the trajectories. I'm almost done with it.

SylvainBertrand, 2016-03-03:
Alright so I just pushed a bugfix. Checkout the branch bugFix/badArmTrajectoryPacketHandling and let me know if that fixes your issue or if have any problem with.

iamwolf, 2016-03-03:
Thanks for looking into this so quickly and the explanation - learned a lot. Will test this in the morning and report back.

iamwolf, 2016-03-04:
@SylvainBertrand we've been trying to test your changes on the robot this morning. Unfortunately the IMU wanders that badly that we cannot use the WholeBodyTrajectory API endpoint at the moment so we switched to using the ArmTrajectory one instead. Moving only one joint on one arm (wristRoll or wristPitch) we managed to trigger an E-Stop due to a massive current spike across most joints.

I've managed to produce a minimally working reproducible sample rosbag, cf. [here](https://drive.google.com/file/d/0B_k3bV-5JKanWDJxdGk0aGg2TFk/view?usp=sharing). Just start the ROSAPISimulator and play back this one. You can see the qdd_ds jerk when it only meant to move one joint slightly with low velocity.

SylvainBertrand, 2016-03-04:
I'm surprised it even does something as at the moment the controller does not support the forearm joints.

iamwolf, 2016-03-14:
The second reported issue went away along with #22. We've tested Sylvain's bug fix for the first and this issue can be closed when #19 is rebased and merged in develop.

closed 2016-03-14 by iamwolf