ihmcrobotics/ihmc-open-robotics-software #20: ROS API has a more seamless rosbag workflow

dljsjr, 2016-03-07:
CC'ing @iamwolf @jlack1987 @SylvainBertrand @mithrandir123 @kodyensley on this for transparency and profit.

Working with rosbags is currently a pain in the ass for us (IHMC) and we should resolve this somehow, so I'm opening up a discussion issue to bat ideas around. If we're going to support a ROS API then we need to support rosbags.

The rosbag problem is many-faceted:

1. Transform information is lacking, which means that rosbags are useless when interacting with the real robot as the location of "world" will be poorly defined and change in between runs,
2. Many IHMC engineers use Windows and OS X so don't have a local ROS installation (yay rosjava), and
3. rosjava does not have an implementation of the rosbag API, which compounds number 2 from above.

The current and short-term workaround is to only allow for using rosbags that are recorded from a simulation and only play them back on simulations, and then either force non Linux devs to have virtual machines or companion computers running Ubuntu that can consume the rosbags.

Unfortunately I have no idea how to really fix problem number 1, and I'm open to suggestions from people that are smarter than I am and know ROS much better than I am. While the proposed solutions from above for fixing 2 and 3 are viable, they also kinda suck. I'm pretty sure we're gonna have to end up writing some code to fix this. So if anybody has any thoughts on this, dump them here.

dljsjr, 2016-03-07:
In the short term, I'm looking in to the rosbag API docs to see if I can just write a rosbag parser in Java. This fixes 2 and 3 but not 1.

For posterity: http://wiki.ros.org/Bags/Format

mauricefallon, 2016-03-07:
This is likely to be relevant:
https://github.com/RobotLocomotion/drake/issues/1733

dljsjr, 2016-03-07:
@mauricefallon Thanks for the link. That's an interesting approach. Sending/receiving messages without the whole ROS stack isn't a problem for us, rosjava actually provides that. It's just that we can't play back rosbags via rosjava and we lose too much information about the world frame, which breaks any of our API endpoints that are control expressed in world frame.

liangfok, 2016-04-15:
Just saw this issue linked from RobotLocomotion/drake#1733, which I've recently begun work on.

I definitely agree with problems 2 and 3, though I believe 3 should be relatively easy to resolve since the binary data format is [known](http://wiki.ros.org/Bags/Format#Formats) meaning a version of `rosbag` can be written in Java.

Regarding problem 1: Since `rosbag` simply records whatever is published on the topics you specify, I don't think `rosbag` is the issue. Instead, the issue appears to lie with the robot's model. If the model includes a 6-DOF floating joint that connects the robot's root link to a static location in the world, the `tf Broadcaster` will publish the transforms for this floating joint on `/tf`, and `rosbag` should dutifully record it. Perhaps the difficulty lies in the ability to sense the robot's current location in the world to begin with? That is certainly difficult and usually requires advanced sensors and algorithms like motion capture systems, SLAM, odometry, etc.

dljsjr, 2016-04-15:
@liangfok You are correct, what I was alluding to was the fact that "world" is not in the same place every time you turn the robot on unless you do some sort of SLAM at boot-up, but even then the notion of "world" is only relevant to that particular map. If the folks at Edinburgh sent us a rosbag, even if we had a notion of a ground truth world reference frame, it would only be applicable to their lab. If we played it back on our robot in our lab it wouldn't make a lot of sense.

liangfok, 2016-04-15:
Wow, yeah, the ability to reproduce a world's coordinate frame across labs is a **huge** problem. One idea is to create a standard "workcell" with exactly the same dimensions that both labs construct, and in which the robots operate. 

To support simulation, an SDF file will need to be created that perfectly captures the dimensions and layout of the standard workcell. The SDF's world frame can be configured to match the workcell's world frame. 

Is the above a valid solution to problem 1?

dljsjr, 2016-04-15:
It's definitely a start. I could see it getting tricky when you want to debug non-trivial stuff like walking over custom terrains and obstacle courses and manipulation tasks but I suppose that could just be a required attachment for all bug reports that contain a rosbag; must have an SDF defining the map that the run happened in. Not sure how easy that would be for folks to generate, though.

dljsjr, 2016-04-15:
I should rename this ticket to reflect that this is specifically in regards to using rosbags as a form of submitting bugs to be reproduced.

liangfok, 2016-04-15:
Ah, yes. I didn't know this was in the context of allowing others to submit reproducible bug reports. Yes, I would say that if the robot is operating in an unstructured environment, it is absolute essential to include a description of the world and how it changes over time with the bug report. Unfortunately, collecting that data from a real execution (as opposed to simulation) would require a crazy amount of sensing. For example, if Atlas is walking over some rubble, the world model will need to capture every small piece of rubble and shift in the rubble over time. Not sure that is realistic given current sensing technology. Even the most expensive LIDAR sensors may not provide the necessary fidelity to capture the changes needed to reproduce the world.

liangfok, 2016-04-15:
Thinking about this some more, if we simulate Atlas walking over rubble in Gazebo and capture ground-truth information about how the world changes (e.g., tiny shifts in the rubble as Atlas walks over it) over time using `rosbag`, we will not have sufficient control authority to replicate these in the real-life "workcell". Thus, the simulation will only be valid if it is physically accurate. Sounds like this is a problem that requires additional basic research to resolve.

dljsjr, 2016-04-15:
@liangfok Most likely. Our preferred method for diagnosing issues is to use our own open source logging software but some people prefer rosbags. If an issue can be reproduced in our simulator then rosbags are fine, it's only an issue with the real robot. This ticket will probably stay open for a loooooooong time :)

wxmerkt, 2016-05-04:
@dljsjr this might come in handy: https://github.com/swri-robotics/bag-reader-java

dljsjr, 2016-05-04:
That's fantastic, thanks Wolfgang. I'll experiment with it very soon.

closed 2018-08-01 by dljsjr