ihmcrobotics/ihmc-open-robotics-software #22: WholeBodyTrajectory ROS API: Short whole-body plan results in fall

iamwolf, 2016-03-11:
We are trying to test the bugFix/badArmTrajectoryPacket branch (as in PR #19) at the moment. When executing a trajectory that is only raising the pelvis z value of the robot issued through the WholeBodyTrajectory ROS API, the robot e-stops. Initial analysis showed a spike in the qdd, e.g. very prominent for the shoulder joints.

A cropped SCS log can be found [here](http://terminator.robots.inf.ed.ac.uk/public/scs/	20160311_183854_ValkyrieRosControlController_ShortWBPlanResultsInFall.tar.gz).

@SylvainBertrand, could you please have a look and let us know whether we can provide any other information?

Thanks!

SylvainBertrand, 2016-03-11:
Thanks for log,  I'm downloading it.

mauricefallon, 2016-03-11:
The plan has 20 samples which raise the pelvis-z by about 1cm in total.

the time_to_waypoints are something like:
1.0 second
1.1 second
1.2
1.3
.....

i.e. a long time to execute the first sample (to avoid the starting spike you had suggested might be the issue), followed by 0.1 seconds to hit every sample after that.

SylvainBertrand, 2016-03-11:
So basically in that run you wanted to command only the pelvis, right?

mauricefallon, 2016-03-11:
yes, just to raise the pelvis but the plan was sent as a WBP BUT with a series of arm and chest way points that were identical in the message

SylvainBertrand, 2016-03-11:
For the arm joints, I see you sent 1 waypoint per joint.
The time to reach the first is 0.0168 sec, and the delta between the initial position and the first waypoint position is around 0.05 rad causing the trajectory to accelerate like crazy and causing the big spikes in desired accelerations.
Did you send the current joint angles for the first waypoint?

SylvainBertrand, 2016-03-11:
I'm gonna look at the rest of the body and try to figure out if we are getting the data from the packet properly.

SylvainBertrand, 2016-03-11:
Alright so the pelvis seems to be fine as the time to reach the first waypoint is set to 1.0168 seconds.
I feel like there was some timing issue for the arm joints and that instead of having the first waypoint at 1.0168 seconds it was set to 0.0168 second.

mauricefallon, 2016-03-11:
Yeah, I think I see the issue on our end. I didn't make the change to insert the 1second offset in all sub components of the WBT message. there is a field called time_from_start in the arm plan and I only changed the place in the pelvis trajectory.

(But, I think the ROS message designing could be improved)

SylvainBertrand, 2016-03-11:
Yeah I don't really like either the current API we're providing you at the moment, I've implemented a new controller API that should be much nicer and Doug is currently working on the ROS API.

dljsjr, 2016-03-11:
I'm hoping to have the new API completed today so we can test it for a few days and release it next week.

mauricefallon, 2016-03-14:
@SylvainBertrand : You were right. Fixing the time offset for the first plan seems to be just fine. The arm jerks at the start of the plans has gone away.

@dljsjr : glad to hear the new API is coming. We'd like to stick with the current one for a day or two so we know that current version is bug free. (Basically we will do do some more arm motions and walking plans for testing.)

iamwolf, 2016-03-14:
Closing since this was a non-issue and was resolved on our end.

closed 2016-03-14 by iamwolf