ihmcrobotics/ihmc-open-robotics-software #23: Valkyrie Forearm Control

wxmerkt, 2016-03-17:
Hi!
We'd like to know what's the status of controlling the position-controlled forearm (not finger!) joints in the whole body controller - is that already supported and executed?

Thanks!

dljsjr, 2016-03-17:
Hey @wxmerkt,

Our current timeline looks something like having the neck working some time next week and to have the forearms working before the next round of robots ship to Boston at the end of the month. The neck stuff is fairly simple as we simply need to modify our ROS control adapter (which I'm working on today), but the forearms require changes to the controller as we don't currently have anything in place that allows for arm control that is partially torque control and partially position control.

wxmerkt, 2016-03-17:
Hey Doug!
Thanks for the quick response. Would it be possible to have a flag to deactivate the neck control? E.g. via a ROS service call. For some experiments we'd like to have our own neck controller in place.

Thanks :)

dljsjr, 2016-03-17:
That's a good question. I'm not opposed to it, I'm just not 100% sure what the implementation details/challenges in doing that would be. Probably a question for @SylvainBertrand. If it's easy for Sylvain to toggle neck control on and off in the controller then it's easy enough for me to make an API endpoint for it.

wxmerkt, 2016-03-22:
Good morning!
Any news or updates on whether it will be toggle-able (the neck's mass and thus its momentum isn't that significant so it should be possible to toggle it?)? Is the ETA still going to be sometime this week? And, as far for the forearms - will that be ready mid next week?

Thanks,
Wolfgang

dljsjr, 2016-03-22:
You can track the progress of the neck stuff on https://github.com/ihmcrobotics/ihmc-open-robotics-software/tree/feature/ROSI-373-valkyrie-position-controlled-neck

Sylvain and I are reviewing my changes today so we should be able to figure out if the neck control will be toggle-able some time today.

Forearm control will use the same code, so once that branch merges in then it's all about augmenting the control algorithm.

wxmerkt, 2016-03-31:
Hi @dljsjr 
What's the current state of this? I assume ROSI-373 has been merged given that the feature branch has been deleted (it's hard to follow changes to "develop" without public PRs - it's the main branch of a monolithic repository and thus there's a lot of activity). Did you figure out whether it's toggle-able?

Thanks :)

dljsjr, 2016-03-31:
Hey @wxmerkt,

Yes, it was merged here: https://github.com/ihmcrobotics/ihmc-open-robotics-software/commit/a75c28fe8783b8fd0fc3dc763c4489670b5a845a

The commit messages for this aren't useful since it's using the pull request number on our Stash instance instead of GitHub. We haven't decided how to fix this problem yet. We don't develop directly against GitHub, it just acts as the public facing portal, so we may have to change the way we do pull requests. I agree that it needs to be more informative. Might just move to a workflow where all of the Pull Requests happen on GitHub.

Making the control "toggle-able" is actually related to the discussion occurring over at https://github.com/NASA-JSC-Robotics/valkyrie/issues/20. Due to the nature of ROS Control, we'd have to rewrite a significant portion of the code that interfaces with the JSC API as well as the controller in order to establish a notion of which joints we have control over.

However, with the new control architecture and API that will be coming out in beta next week, it is our intent to provide both a joint-space and task-space mode for controlling various joints. Would it be sufficient to re-implement your controller and have it talk to our API and send joint space commands to the neck?

closed 2016-07-06 by dljsjr