ihmcrobotics/ihmc-open-robotics-software #24: have high level behavior of standing be reported when fully finished walking

mauricefallon, 2016-03-29:
the high level state of the controller is reported as an integer in:
ihmc_ros/valkyrie/output/behavior

when walking this value is reported as WALKING (i.e. 4).
When the robot puts it foot on the ground for the final step in a walking plan this immediately changes to STANDING (i.e. 3). But the robot is still moving: it settles its foot on the ground and sways into the standing position which takes about 3 seconds.

@dljsjr would it be possible to have this state transition happen after the robot has come to a standstill?
Accounting for this is confusing for autonomy (and different to how BDI behaved)

dljsjr, 2016-03-29:
I can't imagine that this would be too difficult… cc @SylvainBertrand

mauricefallon, 2016-03-29:
Related to that is there a simple way to calculate the length of a walking plan?

e.g. it could be:
( swing_time + transfer_time )* num_of_footsteps + initial_sway_time + final_sway_time

We have the first 3 numbers

dljsjr, 2016-03-29:
It's not exposed via ROS but you can look at `CapturePointPlannerParameters::getDoubleSupportInitialTransferDuration()`. For Valkyrie (ValkyrieCapturePointPlannerParameters) it appears that as of the time of this writing, this is set to 2.0 seconds on the hardware/in Gazebo and 1.0 seconds in SCS:

https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/dad71e261b29436e3f16f6cdada9f6044becf29d/Valkyrie/src/us/ihmc/valkyrie/parameters/ValkyrieCapturePointPlannerParameters.java#L21-L25

We're going to fix the behavior reporting in the new controller architecture, so once that gets released you shouldn't need to compute it, but for now there are the numbers.

mauricefallon, 2016-03-31:
Thanks - that's perfect.

Is my equation above correct? i.e. could we depend on this calculation to predict when a walk will end to within -/+ 1 second.

Obviously having direct feedback from the controller would be preferable to doing this ourselves. It's prone to error (as you say there are parameters we don't have access to in the code)

dljsjr, 2016-03-31:
I believe your equation above is correct. Let me know if this doesn't seem to be the case.

As with all changes to the controller, there won't be hot fixes for these issues. They'll be fixed when we release the new controller architecture with the new controller API. We're hoping this will be on the development branch some time next week, and we'll make a stable release of it as soon as it's been tested heavily on real Atlas and Val hardware.

dljsjr, 2016-05-05:
@mauricefallon In the new API you will see that there is now a Walking Status output topic: 

https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/02740b3609d99412e93da06600dbdb4c6b15e9e4/IHMCROSTools/ROSMessagesAndServices/ihmc_msgs/msg/WalkingStatusRosMessage.msg

This should provide the additional information that you're looking for instead of just relying on the reported high level state. Once you start to integrate the new API, let me know if this is doing what you expect it to do.

dljsjr, 2016-09-14:
@mauricefallon Was this issue ever successfully resolved for you via this endpoint?

closed 2017-01-12 by calvertdw