ihmcrobotics/ihmc-open-robotics-software #25: sending an incorrectly formatted message causes the ROS reciever to crash

mauricefallon, 2016-03-30:
sending an incorrectly formatted footstep plan causes the controller to stop responding to all future messages. It also does this silently without any warning or printf.

My guess (extrapolating from the same kind of behaviour in Python) is that the Controller's ROS receiving thread crashes but the controller is still active.

This can be readily replicated in simulation by putting faulty robot_side id into a message. (the usual ids are 0 or 1)

```
footstep_data_list: 
  - 
    robot_side: 1
    location: 
      x: 0.154122320942
      y: -0.125441902765
      z: 0.0927366825572
    orientation: 
      x: 0
      y: 0
      z: 0
      w: 1
    predicted_contact_points: []
    trajectory_type: 1
    swing_height: 0.05
    unique_id: 0
  - 
    robot_side: 2  <------------------------------- should be either zero or one
    location: 
      x: 0.339570214942
      y: 0.133368271576
      z: 0.0928407591907
    orientation: 
      x: 0
      y: 0
      z: 0
      w: 1
    predicted_contact_points: []
    trajectory_type: 1
    swing_height: 0.05
    unique_id: 0
```

mauricefallon, 2016-03-30:
a minimal python file which sends simple footstep plans. Send a '2' to crash the controller i/o:

[send_basic_steps.py.txt](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/196509/send_basic_steps.py.txt)

dljsjr, 2016-03-30:
I'm able to reproduce this, and your suspicion that the crash is occurring on the ROS receiving thread is correct. It's also why you don't see output. When running via the ROS API, exceptions are thrown to a topic instead of the console. The topic never has anything published to it because of said crash.

I'll look in to where it's happening (I have an idea) and I'll see if I can patch it tonight or early tomorrow morning.

closed 2016-03-31 by dljsjr

mauricefallon, 2016-03-31:
I don't think that solves the problem properly. It will take an incorrect input and clamp it to the enum range which will then cause an arbitrary behaviour that will be difficult to interpret - in particular for a user.

Wouldn't it be better to check if the input is outside of the bounds of the enum range and then throw an assertion with a message?

dljsjr, 2016-03-31:
The fix above is an artifact of other more annoying choices that were made elsewhere in the ROS->IHMC packet architecture. The short version is that it's a (surprisingly) non-trivial fix to spit out an error and then just abandon/ignore the message with the way it is currently set up.

I've made an internal ticket related to this and it's going to be addressed more properly in the next week or two when we rearchitect the message translation layer. This was the only thing I could do for now to keep it from crashing.