ihmcrobotics/ihmc-open-robotics-software #26: Only claim required sliderboard channels

wxmerkt, 2016-04-01:
As bespoken in Slack with @dljsjr:

 * Have a boolean DEBUG or TUNING flag that determines whether all sliderboard channels are mapped or just 7 and 8 (for CoM height and smoothing between standing and walking control), i.e. ``if (DEBUG_OR_TUNING) { claim_all_sliders_and_knobs }``

dljsjr, 2016-04-01:
@SylvainBertrand This would be nice for Valkyrie; the robot currently cannot stand without using the WalkingSliderBoard and we grab all the channels since this is usually used for tuning. In the case of Valkyrie it would be nice to allow for other stuff to be on the traditionally more diagnostic channels.

dljsjr, 2016-05-05:
@wxmerkt We're still looking in to automating the control gain transition but we need physical robot access. If the functionality you contributed is satisfying your use cases currently, I can close this issue. Let me know your thoughts.

wxmerkt, 2016-05-05:
Yeah it's sufficient if it is carried through to the new controller
interface. Happy to close this issue :)

closed 2016-05-05 by dljsjr