ihmcrobotics/ihmc-open-robotics-software #28: Specs of the "New API"

wxmerkt, 2016-04-04:
Hi @dljsjr,
Currently, speaking of "current/legacy" and "new" API is super high-level for us so it is hard to figure out what and how the impact will be. 

Hence, would it be possible for us to receive a specification document for the new API? We'd like to see 

- what will change, 
- how it will be designed (what endpoints, what messages, where do they feed to in the controller, what will we get back as feedback), 
- and also have a note on any breaking/behaviour changes. 

Also, if you could give us a heads-up about what changes with the controller during the rewrite (on a functional level) that would help immensely. Getting our hands on a spec document now will also enable us to give feedback early on to save engineer hours later regarding feature/amendment requests.

Thanks!

dljsjr, 2016-04-04:
Good question @wxmerkt. The short version is that almost everything will break at some level, though in many places it will probably be a simple refactor incorporate the new code, as the underlying functionality should be fairly similar.

The new code just got merged in to `develop` over the weekend, and we're planning to have API documentation available over the next few weeks as we implement the ROS adapter for the new API, as we test on real hardware, and after we're done doing the installations in Boston.

All of the currently releases in the 0.7.x series are based on the "legacy" architecture and API, so we would recommend that you continue to use those or use the head of the `master` branch if you encounter problems with the head of the `develop` branch. The first release that incorporates the new API will be tagged 0.8.x and will include a lot of documentation.

At a super high level: The controller itself will still behave the same; a large portion of what has been done is cleanup/modularization of the architecture improvements to the QP solver so that it is easier to incorporate new features as host sites and teams request them. The API is moving to a nearly 100% trajectory based API, where the existing "PosePacket" functionality can be replicated by sending single waypoint straight line trajectories.

At a less high level, we have some slides/road map documents but nitty-gritty documentation is still a work in progress. @SylvainBertrand can get you more information.

dljsjr, 2016-05-05:
@wxmerkt Keep your eyes on this page of our GH Pages site: https://ihmcrobotics.github.io/documentation/30-control/0-Introduction/0-getting-started/

@SylvainBertrand is working on a living document that is embedded as a slide show that provides an overview of the new API and new controller architecture.

dljsjr, 2016-05-05:
This is now moved to https://ihmcrobotics.github.io/documentation/30-control/1-control/0-getting-started/

dljsjr, 2016-07-06:
@wxmerkt If you or @VladimirIvan or @mauricefallon could give us some feedback on the overview slides, we can button up this ticket.

wxmerkt, 2016-07-06:
Thanks @dljsjr. I think the slides are great with regards to the software implementation. Would it possible to add a diagram regarding the QP formulation?

dljsjr, 2016-07-06:
Of course. Paging @SylvainBertrand

georgwi, 2016-09-15:
@wxmerkt I added a block diagram and some text on the website under the slides. Is that what you were looking for? It is more the structure surrounding the QP though. If you are looking for the actual math formulation I put a paper reference there that describes the framework we use with more equations :)

wxmerkt, 2016-09-16:
Fantastic, thank you very much @georgwi - could you perhaps add the rough name the YoVariables for each of the elements in the diagram would take (e.g. ``[joint_name]qdd_d``). The idea is that when we are tracking down issues and debugging behaviour, it's handy to have a reference about which element in the diagram is which variable in the code. Thanks :)

georgwi, 2016-09-16:
@wxmerkt there is a section with yo variable names now - they are not directly associated with the blocks from the diagram above but I marked the desired accelerations that go into the qp and the accelerations that are achieved by the QP considering all other objectives. You can also find the current and desired references for end effectors and joints. They are used in the feedback controllers to compute the desired accelerations. Hope that helps

wxmerkt, 2016-09-17:
Thank you very much :)

closed 2016-09-17 by wxmerkt