ihmcrobotics/ihmc-open-robotics-software #32: PointCloud2 from Multisense SL in SCS

velinddimitrov, 2016-04-13:
Would it be possible to add support for PointCloud2 msg type from the Multisense in the SL simulator?

(Moved from ihmc_valkyrie_ros as suggested by Doug)

dljsjr, 2016-05-05:
Thanks Velin, sorry for the slow response, we were focused on getting the new API up and running.

Okay-ish news: We will do this.

Bad news: Doing it directly in SCS will probably be a low priority fix solely based on what resources we have right now. None of our code uses PointCloud2 data types (we like the flexibility of working with scans directly) and our current priority is to focus on the control algorithm and making the robot as reliable as possible.

Good news: The ROS `laser_geometry` package provides tools for converting `sensor_msgs/LaserScan` in to PointCloud2 types. I might be able to hack together a small utility node that uses `laser_geometry` to listen to the SCS generated topic and convert it, but it'll be a while before it gets done in SCS itself.

velinddimitrov, 2016-05-10:
Thanks for the tip on laser_geometry. I will take a look and if we get something working before you, we will pull request it if you like.

dljsjr, 2016-07-06:
@velinddimitrov I believe I've almost refactored our ROS nodes that we used on Atlas to be robot agnostic if you haven't found a solution to this yourself. The ultimate resolution to this will most likely be providing these ROS nodes as a separate package instead of changing the behavior of the simulation.

dljsjr, 2018-08-01:
Converting the published laser scans from our sim should be doable via https://github.com/ihmcrobotics/ihmc_multisense_tools

closed 2018-08-01 by dljsjr