ihmcrobotics/ihmc-open-robotics-software #33: Max torque deflection

mauricefallon, 2016-04-29:
For the torque deflection model the controller uses very different upper limits on the deflection:
ValkyrieStateEstimatorParameters -> jointAngleMaxDeflection = 0.25
AtlasStateEstimatorParameters -> jointAngleMaxDeflection = 0.1

0.1 (5 degrees) is pretty huge, but 0.25 (15degrees) incredibly high. I don't think it would ever exceed this limit without about 2000Nm of force. I don't think Val can generate that ;)

@SylvainBertrand  increased it from 0.2 to 0.25 in late January. Any idea why?

@manuelli

mauricefallon, 2016-04-29:
See 35c0f40fc20f026ea1c530d68732a4551ec0f9b8

dljsjr, 2016-05-05:
I might be off base, but I believe the TL;DR here is "knob tuning" back in January when we were still trying to get the robot to walk more reliably. And I believe that the massive amount of "deflection" we were seeing between the pelvis and foot during single support actually ended up being more of a calibration issue than a structural issue. @jlack1987 or somebody else from JSC can chime in on that.

We'll revisit these numbers as soon as we have access to robot hardware.

jlack1987, 2016-05-05:
Yeah we recently found nearly an inch error in the lateral position of the feet that's purely due to calibration error. From a high level perspective, the errors caused by this would be indistinguishable from errors caused by deflection. We have some new parts to fix it and have more parts printing and/or on order. Once we get them ready we'll send you the new parts as well as update the videos on the wiki for the new hip calibration procedure.

SylvainBertrand, 2016-05-05:
@mauricefallon Yo! Sorry for answering so late. Indeed that's way too high. I'll retune it when I get my hands on Valkyrie.

closed 2016-07-06 by dljsjr