ihmcrobotics/ihmc-open-robotics-software #35: Left Elbow Tracking

MarcoCar, 2016-05-09:
We are carrying out some simple experiments to asses the tracking performance with Valkyrie. We are having issues with the left elbow joint. Basically we cannot achieve angles greater than -1.4 rad. The issue shows up when we start the robot (the robot is not able to achieve the symmetric pose we command) and when we command arm motions. We cannot straighten the elbow even if we move that joint only. The joint is not mechanically stuck because we can move it manually when the robot is off. When the robot is on we can push the forearm but it immediately goes back to a wrong pose.

This is the wrong pose when we start the robot
[image]

When we tried to move just the left elbow joint we got this
[image]

Alex, our control guy, suggested that there could be an offset on the desired joint acceleration. We are still investigating because it seems other joints can be affected (likely the shoulder pitch).  We'll try to post more details here.
Can you help to trace back to the origin of this problem?

A small log can be found here
http://terminator.robots.inf.ed.ac.uk/public/logs/20160509_100353_left_elbow.zip

dljsjr, 2016-05-09:
Thanks for the log, we'll take a look.

dljsjr, 2016-05-09:
@MarcoCar Is this with the 0.7.x series or the new controller?

cc @SylvainBertrand

MarcoCar, 2016-05-10:
It is the old controller, we didn't switch to the new one yet.

SylvainBertrand, 2016-05-10:
Yo! So I looked at the log you sent and I'm honestly not sure what's going on.
Could you play with the setpoints when the robot is in standprep in the air please? I want to make sure these joints are working properly. You'll find the setpoint variables in SCS in the format: joint name + "StandPrepAngle" (for instance: leftElbowPitchStandPrepAngle)

dljsjr, 2016-05-10:
@MarcoCar That's what I figured I just wanted to make sure. I'd like @jlack1987 to look at this as well to help us diagnose whether this is a software or electromechanical issue.

Are you familiar with how to run our sliderboard controller? That can help diagnose issues such as these. There is a separate roslaunch for the slider board controller but I don't know if it's integrated in to the JSC startup UI.

SylvainBertrand, 2016-05-10:
You could also see if putting the chest in a certain configuration or shifting the ICP affect this issue.

MarcoCar, 2016-05-11:
We did more tests today. Using the standPrep values works ok although both elbows are not able to reach the zero configuration (arm straight), but this is probably the expected behaviour? Running the robot with the walking controller we figured out that the left elbow doesn't get enough torque. The right elbow is much stiffer (see video).

https://dl.dropboxusercontent.com/u/47739875/elbowJointStiffness.mpg

Alex suggested that the proportional gain can be the problem but kp_leftElbowPitch and kp_rightElbowPitch are both set to the same value. Any idea where in the code this difference come from?

dljsjr, 2016-05-11:
There shouldn't be a difference. If the same gain is leading to drastically different response then that indicates an electromechanical issue to me.

jlack1987, 2016-05-11:
I don't necessarily agree. If the actuator is tracking the desired torque then i'm not sure how it could be electromechanical. I'll look at the logs to make sure it's tracking well, but if you don't mind trying to apply a known torque to each elbow and see if they both measure about the same value. We usually use a fish scale to pull at some known distance from the actuator and see if the measured torque is around what we expect it to be. If it's tracking well and that test shows the measured value is close to reality then i'm not sure how it could be a hardware issue right?

dljsjr, 2016-05-11:
@jlack1987 all fair points. But they're still on the legacy controller and an existing robot that had been working well up to this point. So I'm not sure how a software change would have been introduced. Could be bad torque sensing.

jlack1987, 2016-05-11:
Yeah assuming the log shows its tracking well(i'm about to look at it), the only other thing hardware wise that I could think of would be some hardware issue makes it THINK it's achieving the desired torques but it's actually not. This has happened before to us, so until they can verify with the "fish scale experiment" that it's measuring realistic torques then that's still an option.

jlack1987, 2016-05-11:
Another experiment would be to use the sliderboard controller that IHMC provides(or your own if you'd like) to just do simple PD control around a desired position and see if it tracks position well under a simple PD position loop.

jlack1987, 2016-05-11:
Ok sorry to keep spamming this thread, but I just looked at that quick log and the actuator seems to be tracking torque well. Last verification of hardware is to perform the tests to make sure the elbow torques match reality. Let me know the outcome of that as soon as your able to do it. Thanks!

MarcoCar, 2016-05-11:
Thanks for the feedback. I'll attempt the experiment and post the output here. @dljsjr do you know how can I set up the sliderboard to PD control a single joint? I was trying this morning but I didn't find a way. Thank you

dljsjr, 2016-05-11:
@MarcoCar It's a separate controller, you can'd do it from the normal whole body controller's sliderboard. I don't remember if there's a button for having the JSC UI tell Controller Manager to spawn that controller, you may have to SSH in to link and roslaunch it yourself.

If you're still on the legacy stuff everywhere, you'll want to `roslaunch ihmc_ros_control val_sliderboard_control.launch` or something to that effect.

MarcoCar, 2016-05-11:
Great, I'll try that. Thank you

closed 2017-01-11 by dljsjr