ihmcrobotics/ihmc-open-robotics-software #36: The joint: leftHipYaw has not been registered.

allspawj, 2016-05-11:
Running the new-api-beta1 on a valkyrie
"Be a robot"
roslaunch val_deploy val_controller_manager.launch model_file:=model/urdf/valkyrie_C_no_hands.urdf

roslaunch ihmc_valkyrie_ros val_wholebody_control_robot.launch

Controller partially starts, is selectable from remoteValkyrieVisualizer, however cannot select HighLevelState -> walking (it does nothing)

[ INFO] [1462985624.653557850]: Starting JVM with arguments: -Djava.class.path=ValkyrieController.jar -XX:+UseSerialGC -Xmx10g -Xms10g -XX:NewSize=8g -XX:MaxNewSize=8g -XX:CompileThreshold=1000 -verbosegc -Djava.library.path=lib/
Starting Java VM from path  /home/val/valkyrie
Started Java VM: success
[ INFO] [1462985624.949677928]: Partying hard with max memory of: 3720871936
[ INFO] [1462985625.163270628]: Loading robot model from: 'models/val_description/sdf/valkyrie_C.sdf'
[ INFO] [1462985627.890139359]: [INFO] (NetworkParameters.java:38): Looking for network parameters in network parameters file at /home/val/.ihmc/IHMCNetworkParameters.ini
[ INFO] [1462985627.890958536]: [INFO] (NetworkParameters.java:42): Found Network parameters file at /home/val/.ihmc/IHMCNetworkParameters.ini
[ INFO] [1462985627.892397782]: [INFO] (NetworkParameters.java:69): Looking for network parameters in environment variables
[ INFO] [1462985627.892870819]: [INFO] (NetworkParameters.java:70): Environment variables will override entries in the network parameters file.
Attaching native thread 24639 with priority 45 to JVM
[ INFO] [1462985628.565184643]: Looking for forceSensorDefinition leftAnkleRoll
[ INFO] [1462985628.569506447]: Looking for forceSensorDefinition rightAnkleRoll
[ INFO] [1462985630.401027353]: PelvisRotationalStateUpdater: More than 1 IMU sensor, using only the first one: pelvis_pelvisRearImu
[ INFO] [1462985630.473485884]: PelvisIMUBasedLinearStateCalculator: More than 1 IMU sensor, using only the first one: pelvis_pelvisRearImu
Attaching native thread 24685 with priority 45 to JVM
[ INFO] [1462985634.281517648]: Announcing logging session on: name:eth0 (eth0)
[ INFO] [1462985634.326534100]: Trying port 56572
Attaching native thread 24971 with priority 40 to JVM
Attaching native thread 24973 with priority 94 to JVM
[ INFO] [1462985645.759891008]: Setting estimator thread affinity to processor 1
[ERROR] [1462985645.892412199]: Exception in thread "Thread-6" java.lang.RuntimeException: java.lang.RuntimeException: The joint: leftHipYaw has not been registered.
[ERROR] [1462985645.892545157]: 	at us.ihmc.wholeBodyController.DRCControllerThread.run(DRCControllerThread.java:347)
[ERROR] [1462985645.892646076]: 	at us.ihmc.wholeBodyController.concurrent.MultiThreadedRealTimeRobotController$MultiThreadedRobotControlElementRunner.run(MultiThreadedRealTimeRobotController.java:79)
[ERROR] [1462985645.892726211]: 	at us.ihmc.realtime.RealtimeThread.runFromNative(RealtimeThread.java:133)
[ERROR] [1462985645.892911464]: Caused by: java.lang.RuntimeException: The joint: leftHipYaw has not been registered.
[ERROR] [1462985645.893013333]: 	at us.ihmc.commonWalkingControlModules.controllerCore.command.lowLevel.LowLevelOneDoFJointDesiredDataHolder.throwJointNotRegisteredException(LowLevelOneDoFJointDesiredDataHolder.java:409)
[ERROR] [1462985645.893109047]: 	at us.ihmc.commonWalkingControlModules.controllerCore.command.lowLevel.LowLevelOneDoFJointDesiredDataHolder.setDesiredJointTorque(LowLevelOneDoFJointDesiredDataHolder.java:176)
[ERROR] [1462985645.893217997]: 	at us.ihmc.commonWalkingControlModules.controllerCore.command.lowLevel.LowLevelOneDoFJointDesiredDataHolder.setDesiredTorqueFromJoints(LowLevelOneDoFJointDesiredDataHolder.java:123)
[ERROR] [1462985645.893306241]: 	at us.ihmc.wholeBodyController.diagnostics.DiagnosticsWhenHangingController.doControl(DiagnosticsWhenHangingController.java:320)
[ERROR] [1462985645.893392394]: 	at us.ihmc.wholeBodyController.diagnostics.DiagnosticsWhenHangingController.doAction(DiagnosticsWhenHangingController.java:860)
[ERROR] [1462985645.893472100]: 	at us.ihmc.robotics.stateMachines.GenericStateMachine.doAction(GenericStateMachine.java:114)
[ERROR] [1462985645.893562863]: 	at us.ihmc.commonWalkingControlModules.highLevelHumanoidControl.HighLevelHumanoidControllerManager.doControl(HighLevelHumanoidControllerManager.java:172)
[ERROR] [1462985645.893646710]: 	at us.ihmc.simulationconstructionset.robotController.ModularRobotController.doControl(ModularRobotController.java:20)
[ERROR] [1462985645.893724373]: 	at us.ihmc.wholeBodyController.DRCControllerThread.run(DRCControllerThread.java:339)
[ERROR] [1462985645.893799006]: 	... 2 more
[ INFO] [1462985647.791642668]: Accepted client:  java.nio.channels.SocketChannel[connected local=/10.185.0.30:56572 remote=/10.185.0.5:45825]

dljsjr, 2016-05-11:
Thanks @allspawj. We'll try to set up a Gazebo test case that reproduces this.

For other IHMC'ers, here's a sanitized stack trace for copy/pasting in to Eclipse/IntelliJ for stack trace analysis:

```
Exception in thread "Thread-6" java.lang.RuntimeException: java.lang.RuntimeException: The joint: leftHipYaw has not been registered.
    at us.ihmc.wholeBodyController.DRCControllerThread.run(DRCControllerThread.java:347)
    at us.ihmc.wholeBodyController.concurrent.MultiThreadedRealTimeRobotController$MultiThreadedRobotControlElementRunner.run(MultiThreadedRealTimeRobotController.java:79)
    at us.ihmc.realtime.RealtimeThread.runFromNative(RealtimeThread.java:133)
Caused by: java.lang.RuntimeException: The joint: leftHipYaw has not been registered.
    at us.ihmc.commonWalkingControlModules.controllerCore.command.lowLevel.LowLevelOneDoFJointDesiredDataHolder.throwJointNotRegisteredException(LowLevelOneDoFJointDesiredDataHolder.java:409)
    at us.ihmc.commonWalkingControlModules.controllerCore.command.lowLevel.LowLevelOneDoFJointDesiredDataHolder.setDesiredJointTorque(LowLevelOneDoFJointDesiredDataHolder.java:176)
    at us.ihmc.commonWalkingControlModules.controllerCore.command.lowLevel.LowLevelOneDoFJointDesiredDataHolder.setDesiredTorqueFromJoints(LowLevelOneDoFJointDesiredDataHolder.java:123)
    at us.ihmc.wholeBodyController.diagnostics.DiagnosticsWhenHangingController.doControl(DiagnosticsWhenHangingController.java:320)
    at us.ihmc.wholeBodyController.diagnostics.DiagnosticsWhenHangingController.doAction(DiagnosticsWhenHangingController.java:860)
    at us.ihmc.robotics.stateMachines.GenericStateMachine.doAction(GenericStateMachine.java:114)
    at us.ihmc.commonWalkingControlModules.highLevelHumanoidControl.HighLevelHumanoidControllerManager.doControl(HighLevelHumanoidControllerManager.java:172)
    at us.ihmc.simulationconstructionset.robotController.ModularRobotController.doControl(ModularRobotController.java:20)
    at us.ihmc.wholeBodyController.DRCControllerThread.run(DRCControllerThread.java:339)
    ... 2 more
```

dljsjr, 2016-05-11:
@allspawj Silly question, did you also update the code on your Operator Computer where you run the Remote Visualizer in addition to the code running on the robot? That stack trace indicates that you are requesting the Diagnostic Controller and not Walking, which would be possible because the set of HighLevelState enums is different between the old controller and the new. The Remote Visualizer needs to be running the same code.

allspawj, 2016-05-11:
The stacktrace appears without the visualizer running. Minimum steps to reproduce are only running "be a robot" on nasa gui, then the 

roslaunch val_deploy val_controller_manager.launch model_file:=model/urdf/valkyrie_C_no_hands.urdf

roslaunch ihmc_valkyrie_ros val_wholebody_control_robot.launch

The error appears while loading the controller in the second command. 

On the operator computer https://github.com/ihmcrobotics/ihmc-open-robotics-software.git is on tag new-api-beta1, and I did ./gradlew **:Valkyrie:deploy**, but for sanities sake, I will redeploy and restart link/zelda.

I also did make sure to refresh gradle and clean/rebuild in eclipse.

dljsjr, 2016-05-11:
Disregard my previous comment. I've found the issue.

There is a bug in our DiagnosticsWhenHanging controller, and we use that controller to perform the torque offset calculation.

closed 2016-06-10 by allspawj