ihmcrobotics/ihmc-open-robotics-software #38: Robot might be stomping when taking a step

carlosjoan91, 2016-05-11:
When taking a step robot seems to bring down the foot faster than it brings it up, doesn't seem to be enough force to trigger an estop or fault (so it might be normal), but looks like it is using more force than needed.

To replicate:
Either in sim or robot, send a footstep data list with step commands. Test in data file attached was run on the simulator, sending first a footstep in place, then several with forward movement, Z is set to the value of the current foot position (from tf tree) in all cases. 

log file: https://www.dropbox.com/s/9aju9kic1g04df7/maybeStomping.data.data.gz?dl=0

dljsjr, 2016-05-11:
Thanks @carlosjoan91 

Which version of the software was this generated with? 0.7.x or the current beta?

carlosjoan91, 2016-05-11:
Current beta, but have seen the same thing in the real robot with 0.7.x

dljsjr, 2016-05-11:
Cool, thanks!

SylvainBertrand, 2016-06-28:
Let me know if that is still an issue after switching to the new API/controller.
I haven't noticed such a thing so far. Double-check the height of the footsteps you're sending to the controller.

dljsjr, 2016-07-06:
@carlosjoan91 Any update on this since switching to the beta controller?

carlosjoan91, 2016-07-06:
Sorry, didn't see the previous comments, it looks the same to me as far as
I can tell, but since this is not causing any issues (and the frequent e
stopping problem went away after setting the torque offsets) I assume it is
in fact me being too conservative about how it should step (regarding Z
height I am confident we are setting the right one, so far we tested on
flat ground and I am using the same one the feet have from the tf tree), so
I will vote for closing this.

dljsjr, 2016-07-06:
@carlosjoan91 Stomping means one of two things, either the robot thinks it's falling so it speeds up it's swing to "catch" itself, or the footstep height is in the wrong place (either too high or too low). Too high and the robot will place its foot at the requested height, detect that it isn't in contact with the ground, and stomp down until it comes in contact with something. Too low and I think it's self explanatory.

Can you provide a Python script that triggers the event in Sim (rosbag may also work but we're pretty limited to what we can do w/ rosbags)? For whatever reason we are unable to load the dumped simulation data you provided.

carlosjoan91, 2016-07-06:
Tried to grab a video of the robot doing it and she probably took the smoothest steps I've ever seen, so it was definitely fixed somewhere along the way. Closing.

closed 2016-07-06 by carlosjoan91