ihmcrobotics/ihmc-open-robotics-software #39: Yaw bias estimation performs very poorly - when in motion

mauricefallon, 2016-05-13:
@SylvainBertrand : I think the Yaw bias estimation has a number of short comings.

Firstly, we now have a Vicon system and can measure State estimation performance very accurately. I can tell that the yaw bias, for our robot in the following experiment, was essentially constant at -0.0008 rad/sec.

But the IHMC-estimated value **when the robot is moving** is very unstable. I think this video is self explanatory:
https://www.youtube.com/watch?v=CvfuvTd-YJQ
Log of this:
http://terminator.robots.inf.ed.ac.uk/public/logs/20160503_102222_ValkyrieRosControlController_yaw_bias_snippet/

The observations from the experiment are:
1. the bias is well estimated when stationary - to be -0.0008 (but with too much bandwidth). (0-20 seconds)
2. when the arms move the bias becomes really unreliable (at 20 seconds)
3. when walking the robot shifts between double support (Where the bias is really unstable) and single support (where the bias is fixed)

I think there is likely to be one bug covering 2 and 3, which doesn't accounting for the movement of the pelvis when the robot is in motion.

The knock-on result is that when walking the robot's orientation drifts a huge amount.
This is for a full log where the robot is walking about 80% of the time: **10 degrees in 400 seconds!**
[image]

SylvainBertrand, 2016-06-28:
Yo!
So I looked over the yaw drift estimator/compensator and indeed it was behaving pretty poorly.
I updated the code on the develop branch, it seems to be better now, but I'm not sure if it is good enough for you guys. Just let me know.

mauricefallon, 2016-07-14:
Bonsoir,

This is indeed looking much better. I didn't compare with VICON but what I see in the UI is much better.
I see that isIMUDriftYawRateEstimated is now (nearly always) false when walking. 

Although on occasion that variable is true for a few of ticks when switching the standing foot. I think there could be a snakk bug there.

Another thing is that isIMUDriftYawRateEstimated is false when the robot goes into standing at the start of the experiment and is not set true until the first arm or footstep plan is sent.

SylvainBertrand, 2016-07-14:
Hehe bonsoir!

Alright, so I implemented a new module: IMUBiasStateEstimator that I use for estimating, compensating, and debugging bias related to the IMUs in general. I ended up moving the estimation of the drift in yaw in there. I discovered that the previous algorithm I was using in IMUDriftCompensator was pretty dumb and overly complicated. I ended up with something that I think is much simpler, a bit more reliable, and you shouldn't see much of difference in the result but a small improvement, as for instance the bias shouldn't be estimated during walking at all anymore.
It is on DEVELOP

mauricefallon, 2016-07-15:
Thanks that's great.

`It is on the branch develop.'
I think you mean this commit:
https://github.com/ihmcrobotics/ihmc-open-robotics-software/commit/af5b6d6d2a4f622a1da1a9499cd5bb224cdb5b9c
would you mind mentioning a commit in future - so we can keep track of where the fix is.

closed 2016-07-15 by mauricefallon