ihmcrobotics/ihmc-open-robotics-software #40: Robot feet point inwards when going to home position

carlosjoan91, 2016-05-14:
After starting robot, when servo-ing legs (after all rotors have been initialized, and ihmc controller is running), instead of going to the home position, legs go to a position where each foot points inwards (towards each other). This seems to persist even after sending footsteps with a different orientation (i.e. the robot takes the step and twists the foot, but right before finishing the step, foot twists inwards again)

Log (from cold boot): https://www.dropbox.com/s/pe6pcvftocryrde/legsInwardLog.zip?dl=0

dljsjr, 2016-07-06:
@carlosjoan91 I suspect this is related to torque offsets, which I believe we discussed on Slack shortly after you posted this issue. Are you still seeing issues with this?

carlosjoan91, 2016-07-06:
This was indeed fixed with the torque offsets. Closing

closed 2016-07-06 by carlosjoan91