ihmcrobotics/ihmc-open-robotics-software #41: OutOfMemoryError: Direct Buffer Memory when starting val_wholebody_control_robot.launch

allspawj, 2016-06-10:
[ERROR] [1465575403.536153554]: java.lang.OutOfMemoryError: Direct buffer memory
[ERROR] [1465575403.536453683]: 	at java.nio.Bits.reserveMemory(Bits.java:658)
[ERROR] [1465575403.536588590]: 	at java.nio.DirectByteBuffer.<init>(DirectByteBuffer.java:123)
[ERROR] [1465575403.536665159]: 	at java.nio.ByteBuffer.allocateDirect(ByteBuffer.java:306)
[ERROR] [1465575403.536824277]: 	at us.ihmc.multicastLogDataProtocol.MultiClientStreamingDataTCPServer$ClientHandler$1.newInstance(MultiClientStreamingDataTCPServer.java:194)
[ERROR] [1465575403.536917967]: 	at us.ihmc.multicastLogDataProtocol.MultiClientStreamingDataTCPServer$ClientHandler$1.newInstance(MultiClientStreamingDataTCPServer.java:189)
[ERROR] [1465575403.536991759]: 	at us.ihmc.concurrent.ConcurrentRingBuffer.<init>(ConcurrentRingBuffer.java:90)
[ERROR] [1465575403.537073445]: 	at us.ihmc.multicastLogDataProtocol.MultiClientStreamingDataTCPServer$ClientHandler.<init>(MultiClientStreamingDataTCPServer.java:188)
[ERROR] [1465575403.538797140]: 	at us.ihmc.multicastLogDataProtocol.MultiClientStreamingDataTCPServer$ClientHandler.<init>(MultiClientStreamingDataTCPServer.java:172)
[ERROR] [1465575403.538896573]: 	at us.ihmc.multicastLogDataProtocol.MultiClientStreamingDataTCPServer.<init>(MultiClientStreamingDataTCPServer.java:44)
[ERROR] [1465575403.538984550]: 	at us.ihmc.multicastLogDataProtocol.MultiClientStreamingDataTCPServer.<init>(MultiClientStreamingDataTCPServer.java:52)
[ERROR] [1465575403.539060087]: 	at us.ihmc.robotDataCommunication.YoVariableProducer.start(YoVariableProducer.java:89)
[ERROR] [1465575403.540361913]: 	at us.ihmc.robotDataCommunication.YoVariableServer.start(YoVariableServer.java:95)
[ERROR] [1465575403.540690742]: 	at us.ihmc.valkyrieRosControl.ValkyrieRosControlController.init(ValkyrieRosControlController.java:256)
[ERROR] [1465575403.540781285]: 	at us.ihmc.rosControl.IHMCRosControlJavaBridge.initFromNative(IHMCRosControlJavaBridge.java:103)
[ERROR] [1465575403.540938449]: Initializing controller 'ihmc_valkyrie_control_java_bridge' failed

We have also been getting this error when running the controller for Gazebo. Any ideas?

dljsjr, 2016-06-10:
Under what conditions are you running this roslaunch? Is this on the real robot?

allspawj, 2016-06-10:
Yes, this time was on a real robot, however we get the same error when running val_wholebody_control_gazebo.launch as well.

This time was on the real robot, after going through the steps to turn motor power on.

dljsjr, 2016-06-10:
Are you running the stable `0.7.10` tag or the `develop` branch?

allspawj, 2016-06-10:
Develop.

Also the more output from roslaunch val_deploy val_controller_manager.launch model_file:=model/urdf/valkyrie_C.urdf is

[ INFO] [1465576843.866451093]: Starting JVM with arguments: -Djava.class.path=ValkyrieController.jar -XX:+UseSerialGC -Xmx10g -Xms10g -XX:NewSize=8g -XX:MaxNewSize=8g -XX:CompileThreshold=1000 -verbosegc -Djava.library.path=lib/
Starting Java VM from path  /home/val/valkyrie
Started Java VM: success
[ INFO] [1465576844.410765638]: Partying hard with max memory of: 3720871936
[ INFO] [1465576844.724026182]: Loading robot model from: 'models/val_description/sdf/valkyrie_C.sdf'
[ INFO] [1465576848.088884741]: [INFO] (NetworkParameters.java:38): Looking for network parameters in network parameters file at /home/val/.ihmc/IHMCNetworkParameters.ini
[ INFO] [1465576848.090042857]: [INFO] (NetworkParameters.java:42): Found Network parameters file at /home/val/.ihmc/IHMCNetworkParameters.ini
[ INFO] [1465576848.090979768]: [INFO] (NetworkParameters.java:69): Looking for network parameters in environment variables
[ INFO] [1465576848.091823553]: [INFO] (NetworkParameters.java:70): Environment variables will override entries in the network parameters file.
Attaching native thread 31413 with priority 45 to JVM
[ INFO] [1465576848.779076707]: Looking for forceSensorDefinition leftAnkleRoll
[ INFO] [1465576848.782987324]: Looking for forceSensorDefinition rightAnkleRoll
[ INFO] [1465576850.738719763]: PelvisRotationalStateUpdater: More than 1 IMU sensor, using only the first one: pelvis_pelvisRearImu
[ INFO] [1465576850.808571638]: PelvisIMUBasedLinearStateCalculator: More than 1 IMU sensor, using only the first one: pelvis_pelvisRearImu
Attaching native thread 31509 with priority 45 to JVM
[ INFO] [1465576855.216560106]: Announcing logging session on: name:em1 (em1)
[ INFO] [1465576855.277495608]: Trying port 56408
+The error linked before.

The error is triggered with val_wholebody_control_robot.launch and exact output is 
[INFO] [WallTime: 1465576843.698960] Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [WallTime: 1465576843.702311] Loading controller: ihmc_valkyrie_control_java_bridge
[ERROR] [WallTime: 1465576871.456163] Failed to load ihmc_valkyrie_control_java_bridge
[INFO] [WallTime: 1465576871.457211] Controller Spawner: Loaded controllers: 
[INFO] [WallTime: 1465576871.468258] Started controllers:

dljsjr, 2016-06-10:
RE: Running `develop` on the real robot, we still don't have a robot to test the `develop` branch. @velinddimitrov set something up for us to VPN in, we haven't had a chance to test it yet. My guess is that there are zombie processes or something and that the machine you're trying to run the controller on doesn't have enough RAM available but that may not be the case.

dljsjr, 2016-06-15:
A hat tip to @allspawj for helping me figure this out.

We recently increased the buffer size in our logger: https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/219507337f27a5479fe50858458eec7790e4c09d/RobotDataCommunication/src/us/ihmc/robotDataCommunication/YoVariableDataReceiver.java#L17
The buffer increase is probably not going away any time soon.

This buffer is only used on "real" controllers: controllers that run on real robots. Normally this wouldn't be a problem, except for Valkyrie it's a problem in two ways.

1.) This bug is currently limiting the heap size available to the controller on the physical Valkyrie robot: ihmcrobotics/ihmc-ros-control#1. It only impacts Valkyrie because it is the only robot in which we have to hack the way in which the controller starts so that it can masquerade as a C++ .so and be compatible w/ ROS Control.
2.) When Valkyrie is simulated in Gazebo, it uses the same codepath as the real robot due to the way JSC implemented their API. This in and of itself is kind of cool, but it means that ihmcrobotics/ihmc-ros-control#1 comes back to bite us. It will also be a problem if somebody attempts to run both the controller and the Gazebo simulator on a resource constrained laptop.

Fixing ihmcrobotics/ihmc-ros-control#1 will fix 1.) and 2.) for the most part, but I don't think that you'll ever be able to run our controller + Gazebo on the same machine if it has less than probably 16GB of RAM. Then again, running both Gazebo and our controller at the same time requires a ton of resources as is.

allspawj, 2016-06-15:
For our end, a quick and dirty fix was to decrease the SEND_BUFFER_LENGTH in YoVariableProducer to 4k, and we no longer get that exception, and I was actually able to load the controller on valkyrie.

I did however run into another problem. I was able to do the adapt and transfer torque offsets, however when I put the robot into Walking I get an exception in MotorController. Reran it twice to the same effect, debugging.

Exception in thread "Thread-6" java.lang.RuntimeException: java.lang.NullPointerException
[ERROR] [1466020961.080993246]:         at us.ihmc.wholeBodyController.DRCControllerThread.run(DRCControllerThread.java:347)
[ERROR] [1466020961.081084329]:         at us.ihmc.wholeBodyController.concurrent.MultiThreadedRealTimeRobotController$MultiThreadedRobotControlElementRunner.run(MultiThreadedRealTimeRobotController.java:79)
[ERROR] [1466020961.081156274]:         at us.ihmc.realtime.RealtimeThread.runFromNative(RealtimeThread.java:133)
[ERROR] [1466020961.081327719]: Caused by: java.lang.NullPointerException
[ERROR] [1466020961.081423536]:         at us.ihmc.commonWalkingControlModules.inverseKinematics.JointPrivilegedConfigurationHandler.submitPrivilegedConfigurationCommand(JointPrivilegedConfigurationHandler.java:159)
[ERROR] [1466020961.081527968]:         at us.ihmc.commonWalkingControlModules.momentumBasedController.optimization.MotionQPInputCalculator.updatePrivilegedConfiguration(MotionQPInputCalculator.java:131)
[ERROR] [1466020961.081626582]:         at us.ihmc.commonWalkingControlModules.momentumBasedController.optimization.InverseDynamicsOptimizationControlModule.submitPrivilegedConfigurationCommand(InverseDynamicsOptimizationControlModule.java:276)
[ERROR] [1466020961.081721463]:         at us.ihmc.commonWalkingControlModules.controllerCore.WholeBodyInverseDynamicsSolver.submitInverseDynamicsCommandList(WholeBodyInverseDynamicsSolver.java:216)
[ERROR] [1466020961.081804458]:         at us.ihmc.commonWalkingControlModules.controllerCore.WholeBodyControllerCore.submitControllerCoreCommand(WholeBodyControllerCore.java:87)
[ERROR] [1466020961.081890171]:         at us.ihmc.commonWalkingControlModules.highLevelHumanoidControl.HighLevelHumanoidControllerManager.doControl(HighLevelHumanoidControllerManager.java:174)
[ERROR] [1466020961.081967956]:         at us.ihmc.simulationconstructionset.robotController.ModularRobotController.doControl(ModularRobotController.java:20)
[ERROR] [1466020961.082045026]:         at us.ihmc.wholeBodyController.DRCControllerThread.run(DRCControllerThread.java:339)

dljsjr, 2016-06-15:
@allspawj We are expecting to receive a robot some time very soon, so we'll be able to more easily debug these sorts of errors. Hopefully next week.

In the meantime, can you please file a separate bug with your new stack trace?

closed 2016-09-09 by dljsjr