ihmcrobotics/ihmc-open-robotics-software #43: ROS_MASTER_URI not correctly parsed

mauricefallon, 2016-06-21:
Not explicitly setting the IP address in the ROS_MASTER_URI causes the IHMC API to be unable to hear incoming command messages. (Does not effect outgoing sensor messages)

explicitly setting the IP in the URI currently works:
* export [](ROS_MASTER_URI=http://127.0.0.1:11311)

the default URI that is set when sourcing ROS does not work:
* export ROS_MASTER_URI=http://localhost:11311

dljsjr, 2016-06-21:
Hey Maurice,

We engaged in further discussion about this on Slack, and it unfortunately has nothing to do with whether it is parsed correctly or not (at least, I'm 99% sure of that. I could still be wrong).

The code path for the "parsing" the URIs looks like this: We take the URI that is set, extract the hostname part of it (in this case, `localhost`), then we use Java standard library calls to look up the `InetAddress` representation of the given hostname. ROSJava's API operates on `InetAddress` and not `String` so there's not much else we can do. And if the hostname passed in to create the `InetAddress` doesn't resolve, then the whole thing would error out. We honestly aren't doing that much, and if this is a legitimate bug then it's most likely in the Java standard library or ROSJava. But I may have missed something in my code dive so I won't rule that out.

For posterity, I want to mention that the idea of `localhost` is one of convention, not standardization. There is nothing anywhere that says that `localhost` **must** resolve to `127.0.0.1`; it is resolved via hostname lookup just like `google.com` or `my-host.local` or whatever else may be running through some combination of DNS and your local `hosts` configuration.

For this reason, we highly recommend that you do not leverage `localhost` and rather explicitly use `127.0.0.1` whenever possible. There are too many variables that can interfere with name lookup. At least while I'm still looking in to this.

I realize that this probably isn't a very satisfying response, so I'm not closing the ticket quite yet. I'm going to continue to investigate what could be the issue with this as it still might be some unintended side effect of our refactor even though the physical network and transport layer code didn't change.

I spoke at length with @wxmerkt about this on Slack, and it is clear to me that this issue did not show up until you moved to newer versions of our code while nothing else in your setup changed. So there is a rather obvious smoking gun. I don't have a good explanation for why this would be the case. The networking code and ROS Transport code did not change; only the ROS Message, controller, and API. I will continue to investigate this in an effort to make absolutely sure that we didn't introduce an unintended change to the way the ROS node connections are established.

mauricefallon, 2016-06-21:
Thanks @dljsjr, I think we have discussed this pretty fully.

In summary:

- the new way of querying the ROS IP has changed to use ROS_MASTER_URI instead of your ini file
- ROSJava cannot resolve `localhost` to be `127.0.0.1` for some reason - despite it being the default ROS_MASTER_URI
- Its now necessary to explicitly to export a IP address to ROS_MASTER_URI as a result

I think we'd prefer not to have to do this, but I'm happy for you to declare this a "won't fix" and to close this issue.

dljsjr, 2016-06-21:
@mauricefallon You put it much more succinctly than I did, and you possibly highlighted why the change happened, namely that for a long time we didn't pull the ROS_MASTER_URI from the environment. I'd guess that .ini templates had always contained the IP in them and that's why we've never seen this before.

I'll spend another day or two playing with the code just to make sure this is indeed the issue.

dljsjr, 2016-06-26:
@mauricefallon I've done a little more research on this using a very small self contained example on my local machine, and I was a little bit mistaken on what's going on but I wasn't too far off base. There was another method call that I didn't quite follow through on.

1.) We convert the String that we get either from the .ini file or the environment to a Java `URI` object, which just handles parsing the String and gives easy access to the protocol, host, port, and context.
2.) We store this URI object in a configuration structure
3.) When we start up our ROSJava node, we use the URI object to grab the "host" portion of the URI, and pass that in to `InetAddress.getByName()`, which asks the operating system to perform a name lookup. At this point we're outside of anything we have control over.
4.) Once it's time to start up ROSJava, we actually *do* pass in a String and not an InetAddress. We use `InetAddress::getHostAddress()` to get the String that we pass in to ROSJava.

I traced my way through this on my local computer and the result of all of that is that if your URI is `http://localhost:11311` and your `/etc/hosts` file isn't crazy, nor is your site's DNS, then the resulting String that we send to ROSJava is actually `127.0.0.1`. Which brings me back to my original assumption:

1. Either this is an issue with either ROS and ROSJava where it doesn't see "localhost" and "127.0.0.1" as the same thing, or
2. Something on y'alls network is screwing with the lookup of `localhost`.

An extremely simple self contained example you can run:

```
public static void main(String[] args) throws URISyntaxException, UnknownHostException
   {
      URI testURI = new URI("http://localhost:11311");

      System.out.println("Host: " + testURI.getHost());

      InetAddress byName = InetAddress.getByName(testURI.getHost());

      System.out.println("Inet Address: " + byName);

      System.out.println("Inet Host Address: " + byName.getHostAddress());
   }
```

I'd be curious to see what the results of this are. I believe that the reason we always use IP addresses by having Java make the calls to do the name lookups is because ROSJava, by default, doesn't integrate with your environment variables and has poor support for the "magic" of ROS Networking, and so systems that might use `ROS_IP` and `ROS_HOSTNAME` as a crutch won't always work correctly.

wxmerkt, 2016-06-27:
I've run it on my machine and that's the result:
```
Host: localhost
Inet Address: localhost/127.0.0.1
Inet Host Address: 127.0.0.1
```

dljsjr, 2016-06-27:
Awesome, thanks for doing that @wxmerkt. This confirms my suspicion that something is happening either in ROS itself or in ROSJava where `http://localhost:11311` and `http://127.0.0.1:11311` are for some reason not the same thing. I'm not sure if this is a `ROS_HOST` issue or what.

You could probably make a really strong argument that we (IHMC) shouldn't be converting the hostname to an IP address when we set up our ROS nodes, but I'm a bit baffled as to why either ROS or ROSJava (whichever is misbehaving) would treat those things differently as long as your OS's name lookups are working, unless ROS completely ignores all name lookups outside of ROS_HOSTNAME. I'll spend another day or two on this to see if I can find an appropriate resolution.

dljsjr, 2018-08-01:
We no longer have ROS 1 in our code, we use the ROS bridge, so I'm closing this issue as it has been effectively side-stepped.

closed 2018-08-01 by dljsjr