ihmcrobotics/ihmc-open-robotics-software #45: Valkyrie starting procedure is simplified

SylvainBertrand, 2016-06-26:
Only one command should be needed from the user to:
- Perform the calibration of the joint torque offsets, and foot six axis force sensors
- Start the walking controller as soon as the robot touches the ground.

SylvainBertrand, 2016-06-30:
The starting procedure has been simplified.
The calibration routine and the transition to walking are now automated.
It can be started either from the SCS visualizer:
[image]
or by sending the new ROS message ValkyrieLowLevelControlModeMessage.
Three modes are available: Stand-prep, calibration, and high-level control.

We could not make the transition to walking triggered using the foot force sensor as the robot is too compliant generating not enough force to make it reliable.

dljsjr, 2017-02-07:
Discussed w/ @SylvainBertrand and the current state of the startup procedure is what he intended. Closing for now, and if anybody comes up with any suggestions for further improving startup we can create new tickets.

closed 2017-02-07 by dljsjr