ihmcrobotics/ihmc-open-robotics-software #48: CP overshoot to the right

velinddimitrov, 2016-06-29:
We had an issue with the robot falling when switching from double support to single support w/ the right foot forward. Consistently happens on that side despite spending days to try different combinations of footstep parameters. We haven't been able to confirm on new API due to a msg mismatch (I'm not around this week and can't debug) but we couldn't send footsteps to the robot when I left.

Log: https://drive.google.com/open?id=0ByK9BDWokDuoWW5MX2ZIcEMxWWM

dljsjr, 2016-06-29:
@velinddimitrov Who should I be in contact with in debugging this if you're out? Since we only have the robot for a few weeks I don't want to wait for too long. It's possible that if you pulled develop some time late last night or this morning that there is now a message mismatch but I'm hoping to have a new beta tag out the door today.

velinddimitrov, 2016-06-30:
You can message Murphy. I think she ran some tests yesterday, so it seems the msg mismatch has been fixed.

closed 2023-12-30 by calvertdw