ihmcrobotics/ihmc-open-robotics-software #56: add artificial noise to simulated state estimate

mauricefallon, 2016-07-12:
@dljsjr or @SylvainBertrand: is there a good away of adding drift to the state estimator (in sim)?
The idea would be to change the x, y, z and yaw of the floating base.
I'm guessing you might have a way of turning it on for a unit test.

I have been able to kind of get something to work by adding to rootJointPosition in:
```
PelvisLinearStateUpdater.computePositionFromMergingMeasurements()
```

This causes the estimator to jump forward a few cm on occasion but isn't working right:
```
      FramePoint point = new FramePoint();
      point.set(0.003, -0.003, 0);
      rootJointPosition.add(point);
```

(Thanks a lot for adding back in the localization module @dljsjr , it works just fine!)

SylvainBertrand, 2016-07-12:
We usually corrupt the sensors like the imu but not the state estimator itself. Let me look into that.

dljsjr, 2017-02-07:
After some internal discussion, this is currently a Won't Fix. We believe that this should be achievable by corrupting the IMU enough. We are always opening to revisiting feature requests in the future if there is a technical limitation though.

closed 2017-02-07 by dljsjr