ihmcrobotics/ihmc-open-robotics-software #57: received plan feedback message

edbot, 2016-07-13:
In the old API there was a message called LastReceivedMessage.msg. This was published whenever the controller received a message from us.
It used to be published on /ihmc_ros/valkyrie/output/last_received_message

We used it  for a few things (to check that messages were not being dropped, to sequence together autonomous actions and to test plan tracking)

It seems to have have been removed around about 19d71e380532af1a8d878bbd9537d2e873b45bb3

@dljsjr  : Could you add it back in?

(this is @mauricefallon)

dljsjr, 2016-08-03:
This functionality was disabled for a short amount of time during a refactor on the develop branch but should be working on on the Beta tags:

https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/d9152b4393bd7ab7e7cefeb91609475336cd5aea/DarpaRoboticsChallenge/src/us/ihmc/darpaRoboticsChallenge/ros/RosRobotConfigurationDataPublisher.java#L256

dljsjr, 2016-09-14:
@edbot @mauricefallon Just wanting to confirm that this topic is now working for you before closing this ticket.

closed 2017-01-12 by calvertdw