ihmcrobotics/ihmc-open-robotics-software #58: have valkyrie contact points changed?

mauricefallon, 2016-07-14:
I'm wondering if the contact points for valkyrie's feet have changed - or possibly the urdf containing them? I notice the contact points are not at the corners of the feet, they are now a few cm further forward. Is this intentional?

This is what they look like:
[image]

And Atlas for comparison has them right at the back of the feet:
[image]

FYI: @jlack1987 @dljsjr

jlack1987, 2016-07-14:
I'm not sure if they have changed. I'm 99% sure those come from somewhere in SCS.

SylvainBertrand, 2016-07-14:
I just compared with the contact points from January and they haven't changed.
I agree that they look a bit too conservative on the heels.
We usually refine those by running a few tests on the actual robot which we shipped back to NASA unfortunately. Let me see what I can do.

dljsjr, 2017-02-07:
Updating this to remind myself that it still exists. This needs to be evaluated next time we have access to the physical robot.

closed 2018-08-01 by dljsjr