ihmcrobotics/ihmc-open-robotics-software #59: walking plan not cleared after a fall

mauricefallon, 2016-07-14:
We've been using this new [start up routine](https://github.com/ihmcrobotics/ihmc-open-robotics-software/issues/45). It's awesome!

We've noticed a problem with this sequence of events:
- the robot falls while walking
- we re-calibrate and put the robot into stand-prep
- the robot **immediately** tries to start walking when we command it to go into high-level-control mode.

I think the controller needs to clear the walking plan that was present when it fell

(2GB) IHMC log:
http://terminator.robots.inf.ed.ac.uk/bignas/useful/ihmc/20160714_150849_ValkyrieRosControlController_walking_plan_not_cleared.tar.gz

SylvainBertrand, 2016-07-14:
Glad you like it.
I can't download the log though, it's asking for credentials which I don't know.

SylvainBertrand, 2016-07-14:
I've fixed that bug on the July 1st in the commit (8bb5a3610d771fa0d796f47513e551ef49e85b2a) "Flush commands when the robot falls."

closed 2016-07-15 by mauricefallon