ihmcrobotics/ihmc-open-robotics-software #61: Whole body trajectory ignores arm trajectories during execution

VladimirIvan, 2016-07-25:
When commanding Valkyrie using the whole body trajectory message, it is possible to get the robot into a state where one (or both) of the arms won't execute the trajectory. The arm would remain at it's original position while the rest of the body track the sent whole body commands.

There is no exception or error message printed.

This happens on both the real robot and in the simulator.
We are running off new-api-beta3 tag.

To reproduce this error on a machine with the rostopic tool, start the simulator with the ROS interface.
Download these two files: [nominal.txt](https://www.dropbox.com/s/lbwk4ohh9csin6e/nominal.txt?dl=0), [reach.txt](https://www.dropbox.com/s/y0wbaxeaqdk03fu/reach.txt?dl=0).
When the robot settles at the initial position, execute the nominal trajectory:
```
rostopic pub /ihmc_ros/valkyrie/control/whole_body_trajectory ihmc_msgs/WholeBodyTrajectoryRosMessage -f nominal.txt
```
When the robot reaches the pose, execute the reaching trajectory:
```
rostopic pub /ihmc_ros/valkyrie/control/whole_body_trajectory ihmc_msgs/WholeBodyTrajectoryRosMessage -f reach.txt
```
The robot is supposed to reach backwards with the left arm.

We have seen the robot to get into this state and not being able execute commands unless the controller was restarted.

The arm still moves in the calibration mode after this happens. There are no hardware issues with arm (e.g. faults, estop, ...).

I have investigated the PacketValidityChecker. Every plan passed the check.

VladimirIvan, 2016-07-26:
I have found the reason why the the execution is aborted and why nothing is being reported.

The HandControlModule checks for joint limits but mutes the result and skips execution (per arm).
The following code adds a printed message when that happens:
```java
   public void handleArmTrajectoryCommand(ArmTrajectoryCommand command)
   {
      boolean success;

      switch (command.getExecutionMode())
      {
      case OVERRIDE:
         boolean initializeToCurrent = stateMachine.getCurrentStateEnum() != HandControlMode.JOINTSPACE;
         success = jointspaceControlState.handleArmTrajectoryCommand(command, initializeToCurrent);
         if (success)
            requestedState.set(jointspaceControlState.getStateEnum());
         else
        	 System.err.println("Dropping arm trajectory "+command.getRobotSide());
         return;
      case QUEUE:
         success = jointspaceControlState.queueArmTrajectoryCommand(command);
         if (!success)
            holdPositionInJointspace();
         else
        	 System.err.println("Dropping arm trajectory "+command.getRobotSide());
         return;
      default:
         PrintTools.warn(this, "Unknown " + ExecutionMode.class.getSimpleName() + " value: " + command.getExecutionMode() + ". Command ignored.");
         return;
      }
   }
```

Even with the fix, this is not a desired behaviour for us. Part of the whole body message can't be executed, the whole message should be discarded and a failure status should be sent back to the author.

This is normally handled by action services in ROS.

Could we make this a feature request (especially reporting failure to execute trajectories and aborting the whole execution)?

dljsjr, 2016-08-03:
You are a Grade A detective, @VladimirIvan.

@SylvainBertrand is currently out of town, error reporting is easy enough to fix but I want to wait for him to return to talk to him about the behavior of the control module.

SylvainBertrand, 2016-08-17:
You'll find your output/warning suggestion in the commit dcea4f5970dec23b0157b1f3084159c94f46d2be.

SylvainBertrand, 2016-08-17:
"Could we make this a feature request (especially reporting failure to execute trajectories and aborting the whole execution)?"
This is not straightforward as the message validation happening at the entry of the controller is pretty low-level and has no idea about joint limits, for instance. I'll see what I can do about that.

closed 2018-08-01 by dljsjr