ihmcrobotics/ihmc-open-robotics-software #62: Lidar scan message renamed

simalpha, 2016-07-26:
@dljsjr: The message containing the data from the laser has been renamed from /multisense/lidar_scan to /multisense/filtered_cloud. Was this update intentional?

dljsjr, 2016-08-03:
@simalpha Not really. That's the name of the topic we subscribe to after we run some of our own filtering stuff on the real robot but it probably shouldn't be the topic we publish to in simulation.

dljsjr, 2016-09-09:
@simalpha Just a heads up, I figured out what was going on with this. I will be introducing some commits that revert the behavior to the original topic (`/multisense/lidar_scan`) instead of the `filtered_cloud` topic so that it is consistent with other robots. So by the time the next beta rolls around, you can be prepared.

closed 2016-09-09 by dljsjr