ihmcrobotics/ihmc-open-robotics-software #63: Invalid neck trajectory message crashes the controller

VladimirIvan, 2016-07-26:
When the neck trajectory gets checked for joint limits (and number of commanded joints), the result of this check stops the trajectory waypoints to be passed into the corresponding trajectory generators.
The execution will not stop though. The controller will still attempt to execute the trajectory. This will call the generators and fail with exception ```"Trajectory has no waypoints."```

The fix is to edit the following two functions:
  

1. JointspaceHeadControlState.java

  ```java
   public Boolean handleNeckTrajectoryCommand(NeckTrajectoryCommand command, double[] initialJointPositions)
   {
      if (!ControllerCommandValidationTools.checkNeckTrajectoryCommand(jointsOriginal, command))
      {
    	 if(this.jointsOriginal.length!=command.getNumberOfJoints())
    		 System.err.println("Wrong number of neck joints: expected "+this.jointsOriginal.length+" got "+command.getNumberOfJoints());
    	 else
    		 System.err.println("Neck comamnd out of joint limits!");
         return false;
      }
      
      int numberOfJoints = this.jointsOriginal.length;

      for (int jointIndex = 0; jointIndex < numberOfJoints; jointIndex++)
      {
         OneDoFJoint joint = jointsOriginal[jointIndex];
         MultipleWaypointsTrajectoryGenerator trajectoryGenerator = jointTrajectoryGenerators.get(joint);
         trajectoryGenerator.clear();

         if (command.getJointTrajectoryPoint(jointIndex, 0).getTime() > 1.0e-5)
         {
            trajectoryGenerator.appendWaypoint(0.0, initialJointPositions[jointIndex], 0.0);
         }
         
         SimpleTrajectoryPoint1DList jointTrajectory = command.getJointTrajectoryPointList(jointIndex);
         trajectoryGenerator.appendWaypoints(jointTrajectory);
         trajectoryGenerator.initialize();
      }
      return true;
   }
  ```

1. HeadOrientationManager.java

  ```
   public void handleNeckTrajectoryCommand(NeckTrajectoryCommand command)
   {
      computeDesiredJointPositions(initialJointPositions);
      if(jointspaceHeadControlState.handleNeckTrajectoryCommand(command, initialJointPositions))
      {
    	  requestedState.set(jointspaceHeadControlState.getStateEnum());
      }
   }
  ```

dljsjr, 2016-08-03:
@VladimirIvan Similar to ihmcrobotics/ihmc-open-robotics-software#61 I wanna review this w/ @SylvainBertrand when he gets home.

SylvainBertrand, 2016-08-17:
I'm not really agile with GitHub, the pull request, and all. So it was quicker for me to ignore your pull request, apply your changes locally on my PC, add couple tweaks and push.
It should now be fixed.

closed 2016-09-09 by dljsjr