ihmcrobotics/ihmc-open-robotics-software #65: Calibration offset too conservative/no or low calibration leads to dangerous poses

wxmerkt, 2016-08-15:
As reported on Slack: After upgrading to Daria and ihmc-develop we've noticed that a) the torso was leaning as much backwards as it could such that the pelvis and back were touching, and b) that the default pose had the arms too close to the body, hitting the torso with the elbows when servoing.

With analysis from Vlad and I, we figured out that this is due to the more conservative maximum value for the torque offsets, introduced in https://github.com/ihmcrobotics/ihmc-open-robotics-software/commit/e035e7258cdd96a5c316894b01d08438850fc622. Since our offset file had larger offsets (``~6.x``) stored in them from before the upgrade, the controller would not calibrate at all. The solution was to manually ``0.0`` all values in said file and revert above commit.

**Concrete issues:**

1. [x] Document the script which writes the IHMC offsets into the ``val_description`` coefficients and reset the IHMC offsets in ``~/valkyrie`` -- now available [here](https://gitlab.com/nasa-jsc-robotics/valkyrie/wikis/how-to-write-torque-offsets-to-valkyrie's-actuators) 
2. [ ] The IHMC controller crashes with ``NullPointerExceptions`` when the offset file is not present unter ``~/valkyrie`` -- this should not happen and be handled more gracefully, perhaps via creating the file and defaulting to ``0.0`` if that's safe
3. [ ] Output/warnings if the required torque offset is greater than the maximum allowed. A way to still calibrate, e.g. via an override (which we manually did) - or opening the torque offsets back up.

That's it for now, we'll be in touch regarding a substantive regression in walking performance once we've analysed the logs.

SylvainBertrand, 2016-08-17:
I believe I fixed the NullPointerException in the commit 78b25b086b0b8298e54ac46a839764ef9063b574.

SylvainBertrand, 2016-08-17:
There should now be a warning the first time the maximum torque offset is reached.
When it is the case, you can simply rerun the calibration. The maximum torque offset is not absolute, it just limits how much the offset can be changed per calibration. So running the calibration several times ends up building up the torque offsets with no limits. You don't need an override.

wxmerkt, 2016-08-18:
Thanks Sylvain! I believe this should fix the issues described here. We will follow up once we've tested. Closing this issue for now

closed 2016-08-18 by wxmerkt