ihmcrobotics/ihmc-open-robotics-software #66: SCS Simulation ROS Communication broken in develop

wxmerkt, 2016-08-29:
Using ``new-api-beta3`` works with the ``ROS_IP`` and ``ROS_MASTER_URI`` work-around we've discovered back in June. The current head of develop does not publish ROS communication though - echoing joint_states is empty. Set-up is the same, checking out ``new-api-beta3`` and running it shows traffic, ``develop`` doesnt

dljsjr, 2016-08-29:
@wxmerkt @simalpha Can you post the console output of all related IHMC processes? Stdout, err, exceptions, etc.

dljsjr, 2016-08-29:
From a clean clone of IHMCOpenRoboticsSoftware, on `develop` (longform hash: 7a80a22aa0f19eed0d1d092466dd828f0526019e), running `ValkyrieRosAPISimulator.java`, I am not able to reproduce; I can echo `/ihmc_ros/valkyrie/output/joint_states` without issue and I can send commands to the simulator via the ROS endpoints.

dljsjr, 2016-08-29:
I am unfortunately also unable to reproduce this issue using the `OpenHumanoidsSimulator.java` entry point.

wxmerkt, 2016-08-30:
Cannot reproduce with anymore with ``new-api-beta4``, closing

closed 2016-08-30 by wxmerkt