ihmcrobotics/ihmc-open-robotics-software #67: Yaw correction in localization/pelvis pose correction is fixed

dljsjr, 2016-09-01:
As mentioned in ihmc-open-robotics-software#51, the yaw correction portion of the localization module is disabled because it doesn't work.

This should be fixed.

mauricefallon, 2016-09-02:
This is disappointing. I didn't expect that correcting yaw would be difficult (I probably assumed it worked given that localization video). Your entire control system is setup to use relative goals during walking - which is awesome. (We were stuck with world-positioned goals in MIT which sucked). So this last bit of correcting yaw seems like its a tiny issue given all of that.

This is issue is important to us as we will be working on localization going forward. It also not difficult for us to feedback what we are doing to you or the other Valkyrie teams - for example we are using ETH's libpointmatcher.

dljsjr, 2016-09-02:
Yeah it's important for us to get it to work properly as well, we've been in active discussion the past few days on what needs to go in to getting that code working. From what I've gathered by talking to people, the big hurdle seems to be that the folks who developed that software are no longer here and it is surprisingly more complicated than it would seem to make the external localizer interact with the state estimator in a way that doesn't cause the whole thing to blow up (exacerbated by the fact that pose corrections come from the past, temporally).

I'm not sure if yaw correction ever worked *really* well, even in that localization video we posted on YouTube way back on the old Atlas (which used a different localization module than the one we currently have). The Atlas IMU was just really really good so we always mostly trusted it. And we didn't use any external localization at all during the DRC finals.

So there's a lot of technical debt and bugs to pay down, it's not like we're repairing a system that malfunctioned; it's a system that was never 100% finished to begin with. It will unfortunately take some time.

closed 2017-02-07 by dljsjr