ihmcrobotics/ihmc-open-robotics-software #68: IMU Bias Estimation isn't work

edbot, 2016-09-07:
I see that in July @SylvainBertrand added a module to estimate all the sensor biases rather than just the yaw bias directly. It isn't working for us and the robot rotates continuously when standing still at 2.5 degrees per minute. (What existed previously worked almost entirely perfectly.)

This is the exciting plot of the yaw drifting :)
[image]

Video:
http://terminator.robots.inf.ed.ac.uk/bignas/useful/ihmc/20160830_YawDriftWhenStatic.m4v
20 second log snippet:
http://terminator.robots.inf.ed.ac.uk/bignas/useful/ihmc/20160830_YawDriftWhenStatic_20seconds.tar.gz
140 second log snippet:
http://terminator.robots.inf.ed.ac.uk/bignas/useful/ihmc/20160830_YawDriftWhenStatic_140seconds.tar.gz

This makes things really hard for us. Any ideas how to fix this?

(this is @mauricefallon)

dljsjr, 2016-09-07:
@edbot @mauricefallon Can you provide similar data w/ the bias compensation disabled?

There is a method for determining whether not it is active in ValkyrieStateEstimatorParameters.java:

https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/develop/Valkyrie/src/us/ihmc/valkyrie/parameters/ValkyrieStateEstimatorParameters.java#L211

SylvainBertrand, 2016-09-08:
Sorry about that, I'll look at it

SylvainBertrand, 2016-09-08:
Have you tried to recalibrate the gyro?

mauricefallon, 2016-09-08:
No. We haven't done that since we started using it. do you expect it would get worse with time?
Bare in mind, I think it worked better for your old class.

(would we need to take it from the robot and do it on a bench?)

SylvainBertrand, 2016-09-08:
I'm kinda surprised that the old class worked better.
Nah no need to take it out, just plug in a computer running Windows with the MIP monitor software from Microstrain (https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Configure-Microstrain-IMU's). You just do "Capture Gyro Bias..." in the settings menu once you started streaming data from the IMU (the play button). I personally do that once a day when I'm working on Val.

mauricefallon, 2016-09-09:
Capturing gyro bias through windows everyday would be unworkable. Surely working bias estimation makes that unnecessary.

> I'm kinda surprised that the old class worked better.

The old class would converge to the estimated yaw rate bias and the robot didn't rotate when standing still. It could have been better but it mostly worked.

@SylvainBertrand : Do you need the experiment @dljsjr requested above - with the bias compensation disabled?

SylvainBertrand, 2016-09-09:
I don't need the experiment, I've got all the data I need. I've implemented a new drift compensator that I'll test on Atlas once I'm back at IHMC (I'm currently out of town for another project). I can't reimplement the old one as it didn't work for robot with more than 2 legs.

I did not choose the IMU, and that is what they need, to be calibrated on a regular basis. A bias estimation won't fix an IMU with a bad bias.

dljsjr, 2017-02-07:
@SylvainBertrand fix mentioned above was vetted, we just forgot to update the ticket. Closing for now, but if this issue persists then we can reopen.

closed 2017-02-07 by dljsjr