ihmcrobotics/ihmc-open-robotics-software #69: After a whole body plan, constraint on pelvis is still active

edbot, 2016-09-07:
we've been using the localization module to correct the x,y,z position (with orientation disabled)

We notice that if we 1) stand the robot, and then 2) send corrections, the state estimator moves around  - as we expect. The real robot doesn't do anything

If instead we 1) stand the robot, 2) execute a whole body plan, 3) and then send corrections. The state estimator moves around, as we'd expect. But the physical robot actually moves to (incorrectly) compensate for the correction.

My understanding is that the constraint in the world frame for the pelvis pose (from the WBP) is still active.
@dljsjr : is this understanding correct? How can we un-set the pelvis constraint or should it be unset automatically?

(this is @mauricefallon)

edbot, 2016-09-07:
This is an example:
http://terminator.robots.inf.ed.ac.uk/bignas/useful/ihmc/20160901_155423_ValkyrieRosControlController_PoseCorrectionAfterWholeBodyPlan.tar.gz

1) stand the robot, 
2) execute a whole body plan, (lowering the arms 
3) and then send corrections. 
The state estimator is shifted to the right by 5cm. But the physical robot actually physically leans 5cm to the left to put its pelvis or CoM back in the original X-Y-Z location.

dljsjr, 2016-09-07:
That doesn't *seem* correct to me. cc @SylvainBertrand

SylvainBertrand, 2016-09-07:
Indeed the constraint on the pelvis XY position is held in world. I'll see what I can do about it.

SylvainBertrand, 2016-09-07:
I believe the height should not be affected though.

SylvainBertrand, 2016-09-09:
Any update on this issue?

mauricefallon, 2016-09-11:
@dljsjr can you back port this to new-api-beta4 and we will test it?

... assuming this is a work flow you'd like to use. minor version numbering might help keep track of minor fixes e.g. 0.8.1

dljsjr, 2016-09-12:
@mauricefallon We still haven't made a 0.8.0 release because the code isn't stable enough, so it won't be a hotfix in a traditional sense, but I'll find a way to back port these changes so that you can test them.

dljsjr, 2016-09-12:
@mauricefallon There is now a `new-api-beta5` tag w/ a hotfix branch. Because it doesn't touch the API or ROS library internals. this tag is only on IHMCOpenRoboticsSoftware

mauricefallon, 2016-09-16:
We've verified that this is fixed on Valkyrie with new-api-beta5.

Thanks!

closed 2016-09-16 by dljsjr