ihmcrobotics/ihmc-open-robotics-software #72: CI Build Status is available to the public

dljsjr, 2016-09-13:
The public should be able to view the current status of the CI test suite for IHMC Open Robotics Software. A "build passing" badge in the README might be nice.

There are two obvious options:

1. Fiddle w/ the Robot Lab's internal Bamboo permissions and configuration so that the build status of IHMC Open Robotics Software is available. This would probably involve changing the way the build is done, though, since our internal CI stuff works over the whole codebase including protected and internal projects.

2. Configure this repo to use [Travis](https://travis-ci.org). We'd have to maintain an additional CI config outside of of the Bamboo CI configs, but we wouldn't be duplicating tests on our hardware and the availability of CI builds and status reporting isn't dependent on our facility network being up.

cc @nmertins @calvertdw

dljsjr, 2016-09-13:
FWIW I'm leaning towards Travis.

dljsjr, 2016-09-14:
I spent a few hours yesterday working on Travis-CI integration. It looks promising, but it uncovered some sort of resource leak. This is avoidable if we fragment the Travis build in to a ton of different tasks similar to how we do things on Bamboo, as the leak only seems to happen when a ton of tests run in one build process.

Pros:
-Our stuff will run on Travis

Cons:
- We'll probably get throttled like crazy and our builds will take a really long time
- The .travis.yml file will look ugly

I'd rather fix the resource leak. I'll make a separate issue for this bug.

calvertdw, 2018-07-31:
Option #1 is now done with 5570fb92c830af175836376fe75d813837483675

closed 2018-07-31 by calvertdw