ihmcrobotics/ihmc-open-robotics-software #81: RobotDescription can be generated by reading a URDF

dljsjr, 2016-09-14:


dljsjr, 2017-02-16:
Just updating this, parsing the core of URDF is pretty straight forward but parsing the `<gazebo></gazebo>` extensions is proving to be a little difficult because of the way that the OSRF allows for those tags to contain arbitrary XML that just gets inlined when converting to SDF. Because we're not converting to SDF but actually trying to parse and construct a sim-ready model with sensors from these model descriptions, it's tricky to get JAXB's model of how XML should be formatted with this really weird edge case in URDF.

Solvable, just going to take a while.

dljsjr, 2017-02-18:
Parsing the Gazebo extensions is now (seemingly) fixed in the above referenced commit (fe63e48). Work can resume on generating RobotDescription from URDF.

calvertdw, 2023-12-30:
Done with URDFTools in SCS 2.

closed 2023-12-30 by calvertdw