ihmcrobotics/ihmc-open-robotics-software #88: Pelvis (SE3) refrence trajectory processing?

VladimirIvan, 2016-09-16:
@SylvainBertrand  @georgwi

How do you generate the pelvis (SE3) reference/desired trajectories from the waypoints we send via ROS?
I'm looking at the plots of pelvisDesiredAngularVelocity and pelvisDesiredAngularAcceleration and they are quite jumpy, even though the velocity trajectories I send are smooth.
You can reproduce this in SCS (and ROS) with the following command:

```rostopic pub /ihmc_ros/valkyrie/control/whole_body_trajectory ihmc_msgs/WholeBodyTrajectoryRosMessage -1 -f pelvis_plan.txt```

[pelvis_plan.txt](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/477154/pelvis_plan.txt)

All of this happens in the simulator (but also on the robot).

I've been trying to compute smooth angular velocity trajectories for the pelvis in our software to make the whole body trajectory execution complete. We've been sending zero velocities so far which caused stopping at each waypoint.

SylvainBertrand, 2016-09-16:
The linear part is generated with cubic splines between waypoints. For the orientation part, I used the used the trajectory called "Hermite quaternion curve" introduced in this paper: http://azrael.digipen.edu/MAT351/papers/Kim2.pdf.

dljsjr, 2017-02-07:
Since this is question that was answered with a reference I'm going to close the issue. If there are persistent technical issues related to the splining implementation then by all means please open a new ticket so we can investigate this further.

closed 2017-02-07 by dljsjr