ihmcrobotics/ihmc-open-robotics-software #91: Topic footstep_status only works in simulation and not on real robot

murphywonsick, 2016-10-06:
On the real robot, nothing is publishing to '/ihmc_ros/valkyrie/output/footstep_status', but it works in simulation just fine.

dljsjr, 2016-10-06:
Can you provide more information on which versions of both the JSC software and IHMC software that you're using?

murphywonsick, 2016-10-07:
Our Valkyrie is currently on the master branch for JSC software and the develop branch for IHMC software. However, this hasn't worked for a while now on the real robot. I've never had a problem with it in simulation though and I've been on a few versions.

dljsjr, 2016-10-07:
Which hash on develop though? I still need more information. Our `develop` branch does not track the `master` branch of JSC's code.

If you can provide me either the hash or the tag of all IHMC Software on all relevant computers (link, OCU, and whichever computer is running the API) then I can try to track this down. Currently the only "supported" software for the Daria series of JSC releases are the `new-api-betax` tags. HEAD of `develop` is not supported on Daria.

dljsjr, 2016-10-10:
@murphywonsick If you can get me this info today I can try to get this tested and resolved quickly, as our next release is scheduled for Wednesday.

murphywonsick, 2016-10-10:
Sorry Doug, I've been trying to coordinate with Jordan to get the information for you. He handles all the robot updates and is more frequently with the robot than I am so he is the better person to ask. He should hopefully chime in soon, thanks!

allspawj, 2016-10-10:
Sorry for the delay, I wanted to make sure I could recreate it and see exactly what was going on. 

There are two topics, /ihmc_ros/valkyrie/output/walking_status which does work, and /ihmc_ros/valkyrie/output/footstep_status which does not publish anything on the robot.

ihmc-open-robotics-software is on e3c309a 

ihmc_valkyrie_ros is on 77436ae

The other ros repos are all still synced with develop, as of today.

Is it possible that footstep_status was meant to be deprecated? Technically for our purposes Murphy could switch to using walking status and it should do what she needs, I think.

dljsjr, 2016-10-10:
@allspawj @murphywonsick Footstep Status should notify you when individual steps complete. Walking status should let you know when an entire walking plan has completed.

If you are on `master` of JSC's code, which is currently Daria release, then the only supported releases for IHMCOpenRoboticsSoftware are to use tags or branches with the `new-api<…>` naming scheme as we mentioned in Slack a month or two ago when the branch tracking model diverged. JSC had to introduce some backwards incompatible changes to some of their code that we link against that could lead to strange data making its way in to the IHMC controller. It doesn't appear that your hash is on one of those tags or branches.

You will also need to verify on `link` that the checked out tag for IHMC Ros Control matches the version specified in Valkyrie's gradle file. For example, on the `new-api-beta6` tag: https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/f1deaaacc0f628cb944ef9f6c093868a294f1b71/Valkyrie/build.gradle#L61

I want to note that once JSC releases their upcoming Eeyore release, we are planning to move back to a model where our `develop` branch *does* track their master release. So this should stop being an issue soon.

I recommend using the branch https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/hotfix/new-api-beta5/ or the tag `new-api-beta6`. Let me know if the issue persists on a supported branch.

dljsjr, 2017-01-11:
@allspawj @murphywonsick Is this issue persisting in the latest release of the JSC and IHMC software?

allspawj, 2017-01-11:
We just upgraded to Eyore a bit ago, and val was still stable with our last version of IHMC so I held off on upgrading so people could finish up their projects, but I will get val on the newest correct version of IHMC software today/tomorrow and update the ticket.

allspawj, 2017-01-12:
@dljsjr 
I see that the src is using 0.8.0. The newest ihmc_valkyrie_ros repo wants 0.8.2 which does't actually have a tag. What is the recommended version that I should upgrade valkyrie to?

dljsjr, 2017-01-12:
@allspawj ihmc_valkyrie_ros has a 0.8.2 tag: https://github.com/ihmcrobotics/ihmc_valkyrie_ros/tree/0.8.2

allspawj, 2017-01-12:
Yes, my confusion is that ihmc-open-robotics-software does not, is that not relevant? Should I just be on devel?

dljsjr, 2017-01-12:
@allspawj https://github.com/ihmcrobotics/ihmc-open-robotics-software/tree/0.8.2

allspawj, 2017-01-12:
Upgrade went smoothly, appear to be all set on that end, now running 0.8.2.

The topic /ihmc_ros/valkyrie/output/footstep_status is still not publishing anything though.
/ihmc_ros/valkyrie/controller is registered as a publisher, but no messages are being sent.

A similar topic, /ihmc_ros/valkyrie/output/walking_status, IS receiving messages, so I dont think its a com issue.

dljsjr, 2017-01-12:
Okay. We'll start taking a look at this right away. Thanks for the update.

dljsjr, 2017-01-12:
cc @jlack1987 Probably gonna need your help on this along w/ @allspawj since we don't have access to robot hardware.

jlack1987, 2017-01-12:
No problem, just let me know what you need me to do

velinddimitrov, 2017-01-30:
@dljsjr any update on this? We are running comparisons between simulation and real-hardware, and this bug is preventing us from running the same code on both. Thanks!

dljsjr, 2017-01-31:
@allspawj @velinddimitrov @murphywonsick Sorry about that, we've been bogged down with writing proposals for a little bit.

I'm going to work on getting you a custom build based on 0.8.2 for gathering debug information. Should be available some time today.

velinddimitrov, 2017-01-31:
@dljsjr thanks! let us know if you need anything from us

dljsjr, 2017-02-01:
@velinddimitrov Before I package up the debug build, one last point of clarification: Is this in Gazebo simulation or SCS simulation that you don't receive footstep feedback?

velinddimitrov, 2017-02-01:
@murphywonsick says it works in SCS. Hasn't been tried on Gazebo, but it doesn't work on the real robot.

dljsjr, 2017-02-01:
Sorry I meant which one does it work in not which one does it not work in. Thanks!

dljsjr, 2017-02-01:
@velinddimitrov @allspawj @murphywonsick After spending some time with this I think most of the information I need can actually be gleaned from a robot log. Instead of a build I'm just attaching a patch you can apply to IHMCOpenRoboticsSoftware that enables some debug messages (It's a .txt cuz GitHub won't let me upload it as a .patch for whatever reason). For this to work correctly you'll need to launch the controller from the command line instead of using UI Builder. If you can do a run with the debug output patch applied and then send me a robot log as well as text files containing the output of both the controller and the API node on Zelda, that should give me what I need.

[Output_debug_messages_for_NEU.txt](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/746067/Output_debug_messages_for_NEU.txt)

velinddimitrov, 2017-02-01:
@dljsjr Thanks Doug. Will you be available by Slack on Friday? We will be at NERVE all day to run tests.

dljsjr, 2017-02-06:
@velinddimitrov @murphywonsick @allspawj Can one of you try something for me? I believe that the bug may actually be conflicting versions of a library.

On *both* Link and Zelda, can you delete the contents of `~/valkyrie/lib` and then re-deploy?

velinddimitrov, 2017-02-06:
@dljsjr we will probably not be at NERVE until Friday, @allspawj any chance you could try this out earlier? the debug patch is still applied, so you should pretty quickly be able to see the exceptions related to the msg translation

dljsjr, 2017-02-08:
Finally tracked this down with some help from @allspawj, turns out that the bug was being caused by a method that only exists in Java 8 getting called when converting this status type from IHMC representation to ROS.

jlack1987, 2017-02-09:
For what it's worth I updated the val instructions to now have users install Java 8 instead of Java 7. So if you feel like upgrading to Java 8 those instructions are [here](https://gitlab.com/nasa-jsc-robotics/valkyrie/wikis/install%20java%208)

closed 2017-02-09 by dljsjr