ihmcrobotics/ihmc-open-robotics-software #92: MIDI Sliderboards w/ remote visualizers don't work on macOS

dljsjr, 2016-10-08:
| Required Info ||
|---|---|
|Version or commit hash|a6c6adf30e7cb0ffd423efdd0022d5010c3f8903|
|Sim or Real Robot|N/A|
|Operating System|macOS Sierra|
|JDK|Oracle JDK 1.8.0_102|

####Steps to reproduce issue
Plug a USB MIDI slider board in to a macOS machine, and launch a remote visualizer. Find a YoVariable that is bound to a slider channel, and plot it. Move the slider.

####Expected behavior
The value of the YoVariable should update as the slider moves.

####Actual behavior
Sliderboards seem to "read" correctly, as when using a motorized mixer the channel slides will move in to place at startup. But moving any of the controls on the mixer will not update the YoVariables accordingly.

####Contributors with relevant knowledge or expertise (optional)
@dljsjr @jerrypratt @jespersmith

dljsjr, 2017-02-13:
I've also heard reports that this is true for Windows but I don't have any confirmation yet.

nmertins, 2017-02-15:
I've verified that sliderboards work fine when the server and client are run locally on macOS.

nmertins, 2017-02-15:
I've also run the server in a Linux virtual machine and connected to it with a remote visualizer run on the macOS host machine.

closed 2017-02-15 by nmertins