ihmcrobotics/ihmc-open-robotics-software #93: IntraprocessCommunicationNetwork leaks some type of system resource when many of them are spun up/shut down in a single process

dljsjr, 2016-10-10:
| Required Info ||
|---|---|
|Version or commit hash|Any|
|Sim or Real Robot|N/A, demonstrated in Sim though|
|Operating System|Unverified, but definitely Ubuntu 14.04|
|JDK|Multiple releases of 1.7 and 1.8, both OpenJDK and OracleJDK|

This was found during development of ihmcrobotics/ihmc-open-robotics-software#72

#### Steps to reproduce issue
It's tough to say. Right now it can reliably be triggered by spinning up a Gradle test runner w/ the Include String of `**/Atlas*FastTestSuite.class`. As an example, using the Gradle wrapper in IHMCOpenRoboticsSoftware:

    IS_BAMBOO=true ./gradlew --no-daemon --info test -PremoteDebug=false \
    -PincludeString="**/Atlas*FastTestSuite.class" -PuseClover=false -PlogLevel=verbose \
    -PscsDataBufferSize=8142 -PforkEveryProp=1 -PkeepScsUp=false -PshowScsWindows=false \
    -PshowScsYoGraphics=false -PcreateVideos=false -PawtHeadless=true

#### Expected behavior
The Gradle test runner should run all test suites matching the include pattern

#### Actual behavior
The Gradle test runner eventually hangs. It appears that we are either leaking threads or leaking components of a queue. The amount of time that the tests will run before hanging is proportional to the amount of heap space provided to the test process using the `maxHeapSize` property of the `test` task. Tested with heap sizes in excess of 12gigs, which is an untenable heap size and still not large enough for the test glob to complete.

The leak itself seems to be in an ExecutorService. *At this point we should not rule out a bug in Java*, but I find that highly unlikely. I think we're just doing something wrong w/ the executors. I just have no idea what yet.

#### Contributors with relevant knowledge or expertise (optional)
@dljsjr @jespersmith @calvertdw @btshrewsbury

calvertdw, 2018-07-31:
I would recommend closing this as we have rewritten the intraprocess layer since then and we have JIRA tickets in the DEVOPS backlog for investigating why we must fork so many processes for our tests.

dljsjr, 2018-08-01:
I suspect this issue has been addressed by switching our comms layer too. Closing, if there are other resource leaks I'll file a new issue.

closed 2018-08-01 by dljsjr