ihmcrobotics/ihmc-open-robotics-software #94: Nullptr exceptions when trying to run the diagnostics controller

jlack1987, 2016-10-17:
| Required Info ||
|---|---|
|ihmc_ros_core|d4eb9247ac55376727919a6bafc7a70f15b94fd7|
|ihmc_valkyrie_ros|e8b412141024bb55d466767734472ce5862a4b91|
|ihmcroscontrol|17662efe2adeec6e1bb15e2ed1b372ca86e6703f|

|Sim or Real Robot|Valkyrie real robot|
|Operating System|Ubutnu 14.04|
|JDK|1.7.0.111|

#### Steps to reproduce issue
With the nasa controller manager running, just execute `roslaunch ihmc_valkyrie_ros val_diagnostic_control_robot.launch`

#### Expected behavior
Expected robot to wiggle for a bit then write a diagnostics output file

#### Actual behavior
Whole bunch of nullptr exceptions

[ERROR] [1476733733.691556941]: java.lang.NullPointerException
[ERROR] [1476733733.691655693]: 	at us.ihmc.valkyrieRosControl.ValkyrieRosControlLowLevelController.doControl(ValkyrieRosControlLowLevelController.java:298)
[ERROR] [1476733733.691724333]: 	at us.ihmc.valkyrieRosControl.ValkyrieRosControlSensorReader.writeCommandsToRobot(ValkyrieRosControlSensorReader.java:86)
[ERROR] [1476733733.691796760]: 	at us.ihmc.valkyrieRosControl.automatedDiagnosticControl.ValkyrieAutomatedDiagnosticController.doControl(ValkyrieAutomatedDiagnosticController.java:223)
[ERROR] [1476733733.691867045]: 	at us.ihmc.rosControl.IHMCRosControlJavaBridge.updateFromNative(IHMCRosControlJavaBridge.java:131)

#### Contributors with relevant knowledge or expertise (optional)

dljsjr, 2016-10-17:
@jlack1987 Can you also provide a hash/tag for IHMC Open Robotics Software on the machine where you deployed the controller from?

dljsjr, 2016-10-18:
@jlack1987 Your clones of ros_core and valkyrie_ros seem to be very out of date whereas IHMCOpenRoboticsSoftware and ihmc-ros-control seems on the latest releases. I don't necessarily expect updating them to fix this issue but it does eliminate a lot of moving parts. Please update ihmc_ros_core and ihmc_valkyrie_ros to 0.8.0 and let me know if that changes anything.

dljsjr, 2016-10-18:
@jlack1987 Never mind, I believe I've found the source of the bug. Since we moved the torque offset generation stuff in to the regular controller, the diagnostic controller has rotted. The `commandInputManager` being utilized on line 298 is not null checked because it is guaranteed to exist in the presence of a MomentumBasedControllerFactory but the pathway that we take to embed this component in the low-level controller occurs through the sensorReaderFactory and the sensor reader.

The entry-point to this is `ValkyrieRosControlSensorReaderFactory::attachControllerAPI(CommandInputManager commandInputManager, StatusMessageOutputManager statusOutputManager)`:
https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/70125a1b61ebb32c844e7b4e480340b787dec063/Valkyrie/src/us/ihmc/valkyrieRosControl/ValkyrieRosControlSensorReaderFactory.java#L194

You can see that this gets called in the ValkyrieRosControlController but not in the ValkyrieAutomatedDiagnosticController. So the Diagnostic controller does not have a commandInputManager but we do not null-check around it.

cc @SylvainBertrand

jlack1987, 2016-10-18:
Awesome, thanks for taking a look!

dljsjr, 2017-02-07:
@jlack1987 Was this fixed?

jlack1987, 2017-02-07:
I'm not sure. I don't recall us ever successfully running it though