ihmcrobotics/ihmc-open-robotics-software #98: Adding an API for updateing Online Footstep Planning

alexrobotics, 2016-10-31:
| Required Info ||
|---|---|
|Version or commit hash| release 0.8.0|
|Sim or Real Robot| Both sim and real robot | 
|Operating System| Ubuntu 14.04) |
|JDK| 1.8.0_91 |

#### Expected behavior
To have an API for users to develop online footstep planning to update future 3-steps at every control loop. Goal is to have a reactive walking behavior that the robot can perform push recovery during walking. 

#### Actual behavior
At the moment, once a series of planned footsteps is sent, the existing software architecture only executes, pauses or resumes these pre-planned footsteps till the end.

mauricefallon, 2016-10-31:
I don't think this is true. The behaviour is already implemented

You can replace the footstep plan mid-walk as shown in this video:
https://www.dropbox.com/s/qle1dmw6eomxxly/2015-09-continuous-walking.ogv?dl=0

alexrobotics, 2016-11-01:
@mauricefallon Opening this issue is exactly to follow what Sylvain and us discussed last Friday. The video you show probably does not corresponds to what reactive walking requires. I will generously give them time to update API as they know what need to be done. No more discussion here to confuse people.

SylvainBertrand, 2016-11-03:
Yo Alex!
There might be some confusion in the air.
Alright the first thing I said I would add to the API for you is a new message to adjust a footstep while the robot is already in single support. So you can have a push recovery module living outside the controller. That part is almost done.
For the second thing, you were asking if we had some kind of velocity based footstep generator so the robot is continuously walking and the walking direction can be controlled via a joystick for instance; we do have one. This footstep generator does not include any kind of reactive behavior to recover from disturbances. I will look into enabling the joystick control for Valkyrie, the API will not change for that one.

These are the only two that we said I will do, so you can work on push recovery.

alexrobotics, 2016-11-03:
@SylvainBertrand Hi Sylvain, yes, your description is more accurate and is exactly what we meant during the discussion. Thanks for clarification and looking forward to new features.

SylvainBertrand, 2016-11-03:
Ok good, I was worried for a sec we weren't on the same page.

SylvainBertrand, 2016-11-03:
So to be able to add the new message for doing footstep adjustments, I need you guys to test a patch for me. It is for switching to our swing foot trajectory, I haven't had the chance to test it on Val yet.

To test it, just try to do some flat ground walking, if you don't notice any difference with the patch, it means that it is good.
Here is the patch:
[Switch-to-new-swing-trajectory.zip](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/569872/Switch-to-new-swing-trajectory.zip)

alexrobotics, 2016-11-14:
@SylvainBertrand The patch worked all fine.

dljsjr, 2017-02-09:
Fixed in 2d995eb8ccb6b17bcb1b838f9f507a6cdb2e884f

closed 2017-02-09 by dljsjr