ihmcrobotics/ihmc-open-robotics-software #99: Disable core affinity in ValkyrieAffinity when running in Gazebo environment

dljsjr, 2016-11-08:
| Required Info ||
|---|---|
|Version or commit hash|0.8.0|
|Sim or Real Robot|Valkyrie in Gazebo/SRCSim|
|Operating System|Ubuntu 14.04|
|JDK|N/A|

When running Val in Gazebo, core isolation is not needed because we run in lockstep with the physics loop. It's reasonable to assume that developers won't be using machines that line up with what the real robot hardware looks like. At least one person has run in to a launch problem where they cannot run the simulation software on a 2 physical core computer.

We should use the parameterization added in 292a659d1f6c56173ff76037707150c72b5c3f2d to parameterize ValkyrieAffinity to avoid this.

#### Contributors with relevant knowledge or expertise (optional)
@dljsjr

eschulma, 2016-11-10:
I can confirm that more than one person has this problem.

dljsjr, 2016-11-10:
@eschulma If the build and test suites for these changes pass, they should make it in to a release of our software over the weekend. We'll work w/ OSRF to get this release integrated in to SRCSim next week ASAP.

cc @scpeters

scpeters, 2016-11-10:
I assume you mean using the `IS_GAZEBO` variable? Sounds good to me. I expect it would only require redeploying the valkyrie controller? I can try deploying it from a branch when one is ready then uploading a new `valkyrie_controller.tar.gz` file.

dljsjr, 2016-11-10:
Correct you should only need to redeploy the controller once the release is available.

theredcomputer, 2016-11-19:
Thank you for working to get this rectified so quickly! Will this issue be resolved sometime this coming week?

eschulma, 2016-11-20:
I am also interested in the release status of this -- thanks. Short of buying a new computer, I can't help my team right now.

dan9thsense, 2016-11-22:
Standing by for this release, gone as far as I can with two cores............

dan9thsense, 2016-11-29:
Should we re-install with these changes to get it to work with 2 cores?  What is the best way to install these changes over an existing build?

nmertins, 2016-11-29:
I'm currently writing up a tutorial for how to apply these changes to an existing build. Should be done by the end of the day.

scpeters, 2016-11-29:
I just uploaded a new [valkyrie_controller.tar.gz](http://gazebosim.org/distributions/srcsim/valkyrie_controller.tar.gz) (step 10 in [installation instructions](https://bitbucket.org/osrf/srcsim/wiki/system_setup)), so please delete the `~/valkyrie` folder and download the new one and extract it in its place. This should fix the issues with dual-core CPU's.

closed 2017-01-11 by dljsjr