ihmcrobotics/ihmc-open-robotics-software #102: Logger throws NullPointerException

jlack1987, 2016-12-30:
| - | - |
|---|---|
|Version or commit hash|6b18d59c42bae5a12981a183c452e8bb61a2992f|
|Sim or Real Robot|Real Robot|
|Operating System|Ubuntu 14.04|
|JDK|```java version "1.8.0_111"```|
|Logger Version|0.7.10|

#### Steps to reproduce issue
We just run the logger executable and when a new control session comes up it crashes with the ```NullPointerException```. 

#### Expected behavior
It connects to the logger, logs the data, and dumps the log to file after it's finished.

#### Actual behavior
```bash
val@palantir:~/IHMCLogger-0.7.10/bin$ ./IHMCLogger -s
Starting YoVariableLoggerDispatcher
[INFO] (NetworkParameters.java:38): Looking for network parameters in network parameters file at /home/val/.ihmc/IHMCNetworkParameters.ini
[INFO] (NetworkParameters.java:42): Found Network parameters file at /home/val/.ihmc/IHMCNetworkParameters.ini
[INFO] (NetworkParameters.java:69): Looking for network parameters in environment variables
[INFO] (NetworkParameters.java:70): Environment variables will override entries in the network parameters file.
[INFO] (YoVariableLoggerDispatcher.java:29): Listening on interface name:p5p1 (p5p1)
Client started, waiting for announcements
New control session came online 910571633764871260
Logging sesion 910571633764871260
910571633764871260
Cameras: [0, 1]
Video stream: /239.255.25.2:56320
Requesting handshake
[INFO] (KryoObjectClient.java:130): Success! Connected KryoClient to SCS at ip /10.185.0.10 on port 56322
java.lang.NullPointerException
	at com.google.protobuf.AbstractParser.parseFrom(AbstractParser.java:180)
	at com.google.protobuf.AbstractParser.parseFrom(AbstractParser.java:185)
	at com.google.protobuf.AbstractParser.parseFrom(AbstractParser.java:49)
	at us.ihmc.robotDataCommunication.generated.YoProtoHandshakeProto$YoProtoHandshake.parseFrom(YoProtoHandshakeProto.java:4777)
	at us.ihmc.robotDataCommunication.YoVariableHandshakeParser.parseYoProtoHandshake(YoVariableHandshakeParser.java:61)
	at us.ihmc.robotDataCommunication.YoVariableHandshakeParser.parseFrom(YoVariableHandshakeParser.java:84)
	at us.ihmc.robotDataCommunication.YoVariableClient.start(YoVariableClient.java:204)
	at us.ihmc.robotDataCommunication.logger.YoVariableLogger.<init>(YoVariableLogger.java:47)
	at us.ihmc.robotDataCommunication.logger.YoVariableLoggerDispatcher.logSessionCameOnline(YoVariableLoggerDispatcher.java:59)
	at us.ihmc.multicastLogDataProtocol.broadcast.LogSessionBroadcastClient.handleRequest(LogSessionBroadcastClient.java:132)
	at us.ihmc.multicastLogDataProtocol.broadcast.LogSessionBroadcastClient.run(LogSessionBroadcastClient.java:81)
```

dljsjr, 2017-01-05:
@jlack1987 Which version of the controller are you using? The logger and controller should always be the same version.

You can find the binaries for the logger distributions on our bintray: https://bintray.com/ihmcrobotics/distributions/IHMCLogger

If you're running it in a Docker container on one of the QNAPS then you just need to make sure you're pulling the "latest" tag from Dockerhub. The Logger's Dockerhub repo is here: https://hub.docker.com/r/ihmcrobotics/ihmc-logger-docker/

jlack1987, 2017-01-05:
We aren't running it in a docker. I'll deploy a new one just to be sure the versions are consistent and send an update on how it went.

jlack1987, 2017-01-09:
Looks like you were correct @dljsjr. Will close this ticket, thanks!

closed 2017-01-09 by jlack1987