ihmcrobotics/ihmc-open-robotics-software #116: ROS topic Footstep status prepends bogus fields with NAN values to each message.

allspawj, 2017-02-08:
| Required Info | |
|---|---|
|0.8.2||
|Valkyrie hardware||
|Ubuntu 14.04||
|OpenJDK 1.7.0_121 and  1.8.0_111||

#### Steps to reproduce issue
Using ROS API send a footstep to /ihmc_ros/valkyrie/control/foot_trajectory

#### Expected behavior
The contorller publishes to /ihmc_ros/valkyrie/output/footstep_status with the footstep status for each step.

#### Actual behavior
The message contains contains the proper information, however it also publishes bogus fields with NAN values

```
status: 0
footstep_index: 1
robot_side: 0
actual_foot_position_in_world:
 x: nan
 y: nan
 z: nan
actual_foot_orientation_in_world:
 x: nan
 y: nan
 z: nan
 w: nan
unique_id: 0
---
status: 1
footstep_index: 1
robot_side: 1
actual_foot_position_in_world:
 x: 0.00883457447116
 y: 0.241212263204
 z: 0.101079834617
actual_foot_orientation_in_world:
 x: 0.0353534929723
 y: -0.0169808031996
 z: 0.731665570712
 w: 0.680534551284
unique_id: 0
---
status: 0
footstep_index: 2
robot_side: 0
actual_foot_position_in_world:
 x: nan
 y: nan
 z: nan
actual_foot_orientation_in_world:
 x: nan
 y: nan
 z: nan
 w: nan
unique_id: 0
---
status: 1
footstep_index: 2
robot_side: 0
actual_foot_position_in_world:
 x: -0.280019805439
 y: 0.221660631648
 z: 0.102628189902
actual_foot_orientation_in_world:
 x: -0.00590754583005
 y: -0.0520905396868
 z: 0.730522621612
 w: 0.680873245098
unique_id: 0
```

#### Contributors with relevant knowledge or expertise (optional)
@dljsjr

dljsjr, 2017-02-08:
@allspawj Is it NaN in sim as well?

allspawj, 2017-02-08:
It seems the same behaviour is in sim as well.

dljsjr, 2017-02-08:
Turns out that somebody already found this bug and fixed it. So it should be resolved in the upcoming release.

dljsjr, 2017-02-09:
Fixed in 2d995eb8ccb6b17bcb1b838f9f507a6cdb2e884f

closed 2017-02-09 by dljsjr