ihmcrobotics/ihmc-open-robotics-software #118: ihmc_ros_java_adapter fail resolving dependency on oss-parent.pom

mikaelarguedas, 2017-02-24:
Hello friends,
Pardon my limited knowledge of gradle, maven and dependency resolution, here is what I was able to track down.

TL;DR: when running `valkyrie_warmup_gradle_cache.launch` the dependency chain bubbles down to `oss-parent.pom`. It's searching for it in the [ros_java maven repo](https://github.com/rosjava/rosjava_mvn_repo/). But this pom has been removed in a cleanup PR **yesterday** (see [here](https://github.com/rosjava/rosjava_mvn_repo/pull/20/files#diff-4f89a7720694f245412245296e4c73e6))
So the build fails (more details below)

| Required Info | |
I'm running `roslaunch ihmc_valkyrie_ros valkyrie_warmup_gradle_cache.launch` using the latest [valkyrie_controller](https://osrf-distributions.s3.amazonaws.com/srcsim/valkyrie_controller.0.9-support.20170217.2950d3544fda.tar.gz) released by OSRF. And gradle fails to resolve dependencies.
|---|---|

|Version or commit hash||
I'm using this [gradle_cache tarball](https://osrf-distributions.s3.amazonaws.com/srcsim/gradle_cache.tar.gz)

|Sim or Real Robot||
NA
|Operating System||
Running in a ubuntu14.04 docker container
|JDK||
java-8-openjdk-amd64

#### Steps to reproduce issue

Download the [valkyrie_controller tarball](https://osrf-distributions.s3.amazonaws.com/srcsim/valkyrie_controller.0.9-support.20170217.2950d3544fda.tar.gz) and the  [gradle_cache tarball](https://osrf-distributions.s3.amazonaws.com/srcsim/gradle_cache.tar.gz) on a system with ROS Indigo installed and up to date. Extract them and launch `roslaunch ihmc_valkyrie_ros valkyrie_warmup_gradle_cache.launch`
#### Expected behavior

Resolve all dependencies a build gracefully :tada: 
#### Actual behavior

Fails to resolve dependency on oss-parent.pom
```
> Could not find oss-parent.pom (org.sonatype.oss:oss-parent:7).
   Searched in the following locations:
       https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/sonatype/oss/oss-parent/7/oss-parent-7.pom
```
#### Full Console log
```
:buildSrc:build
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/Valkyrie/0.9.0/Valkyrie-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCCommunication/0.9.0/IHMCCommunication-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCHumanoidRobotics/0.9.0/IHMCHumanoidRobotics-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCSystemIdentification/0.9.0/IHMCSystemIdentification-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCStateEstimation/0.9.0/IHMCStateEstimation-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/CommonWalkingControlModules/0.9.0/CommonWalkingControlModules-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCAvatarInterfaces/0.9.0/IHMCAvatarInterfaces-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCJMonkeyEngineToolkit/0.9.0/IHMCJMonkeyEngineToolkit-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCROSTools/0.9.0/IHMCROSTools-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCRobotDataLogger/0.9.0/IHMCRobotDataLogger-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/ModelFileLoader/0.9.0/ModelFileLoader-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/SensorProcessing/0.9.0/SensorProcessing-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/SimulationConstructionSet/0.9.0/SimulationConstructionSet-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCPerception/0.9.0/IHMCPerception-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCWholeBodyController/0.9.0/IHMCWholeBodyController-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCJavaToolkit/0.9.0/IHMCJavaToolkit-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCRoboticsToolkit/0.9.0/IHMCRoboticsToolkit-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCGraphicsDescription/0.9.0/IHMCGraphicsDescription-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCRobotDescription/0.9.0/IHMCRobotDescription-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCRobotModels/0.9.0/IHMCRobotModels-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCRobotDataVisualizer/0.9.0/IHMCRobotDataVisualizer-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCSimulationToolkit/0.9.0/IHMCSimulationToolkit-0.9.0.pom
Download https://jcenter.bintray.com/org/boofcv/geo/0.24.1/geo-0.24.1.pom
Download https://jcenter.bintray.com/org/georegression/georegression/0.11/georegression-0.11.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCFootstepPlanning/0.9.0/IHMCFootstepPlanning-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/ConvexOptimizationAdapter/0.9.0/ConvexOptimizationAdapter-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCHumanoidBehaviors/0.9.0/IHMCHumanoidBehaviors-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCRobotBehaviors/0.9.0/IHMCRobotBehaviors-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/Vicon/0.9.0/Vicon-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/KalmanProject/0.9.0/KalmanProject-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/ParameterOptimization/0.9.0/ParameterOptimization-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCSwingPlotting/0.9.0/IHMCSwingPlotting-0.9.0.pom
Download http://dl.bintray.com/ihmcrobotics/maven-release/us/ihmc/IHMCGeometry/0.9.0/IHMCGeometry-0.9.0.pom
[... 73 lines cut ...]
                     > Could not resolve org.orbisgis:orbisgis-nexus:1.
                        > Could not resolve org.orbisgis:orbisgis-nexus:1.
                           > Could not parse POM https://jcenter.bintray.com/org/orbisgis/orbisgis-nexus/1/orbisgis-nexus-1.pom
                              > Could not find oss-parent.pom (org.sonatype.oss:oss-parent:7).
                                Searched in the following locations:
                                    https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/sonatype/oss/oss-parent/7/oss-parent-7.pom
                        > Could not resolve org.orbisgis:orbisgis-nexus:1.
                           > Could not parse POM https://repo1.maven.org/maven2/org/orbisgis/orbisgis-nexus/1/orbisgis-nexus-1.pom
                              > Could not find oss-parent.pom (org.sonatype.oss:oss-parent:7).
                                Searched in the following locations:
                                    https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/sonatype/oss/oss-parent/7/oss-parent-7.pom
               > Could not resolve org.orbisgis:poly2tri:0.1.2.
                  > Could not parse POM https://repo1.maven.org/maven2/org/orbisgis/poly2tri/0.1.2/poly2tri-0.1.2.pom
                     > Could not resolve org.orbisgis:orbisgis-nexus:1.
                        > Could not resolve org.orbisgis:orbisgis-nexus:1.
                           > Could not parse POM https://jcenter.bintray.com/org/orbisgis/orbisgis-nexus/1/orbisgis-nexus-1.pom
                              > Could not find oss-parent.pom (org.sonatype.oss:oss-parent:7).
                                Searched in the following locations:
                                    https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/sonatype/oss/oss-parent/7/oss-parent-7.pom
                        > Could not resolve org.orbisgis:orbisgis-nexus:1.
                           > Could not parse POM https://repo1.maven.org/maven2/org/orbisgis/orbisgis-nexus/1/orbisgis-nexus-1.pom
                              > Could not find oss-parent.pom (org.sonatype.oss:oss-parent:7).
                                Searched in the following locations:
                                    https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/sonatype/oss/oss-parent/7/oss-parent-7.pom
```

dljsjr, 2017-02-24:
Ugh.

Yes it seems that they are no longer vendoring a wide variety of third party libraries. Unfortunate that this was pulled out from under us.

Annoyingly, the only way to fix this is by doing *yet another* release of IHMC ROS Core, the same library Jose is in the middle of releasing.

dljsjr, 2017-02-24:
I worked with @mikaelarguedas and @scpeters on this for quite a while and we were not able to reproduce. It seems there was either a network hiccup or some POMs got confused about something, but this issue cannot be repro'd and we were able to generate a build so I am closing.

closed 2017-02-24 by dljsjr

scpeters, 2017-03-13:
So apparently I can reproduce this now, not sure why.

* https://gist.github.com/scpeters/a3444ef7c3faeccc739b12ef4fcd02fa

scpeters, 2017-03-13:
This is not using the 0.9.2 debs of `ihmc_ros_core`. Would that fix this?

dljsjr, 2017-03-13:
I didn't make any changes to ihmc_ros_core that would impact how those things are downloaded. Can you reproduce reliably?

scpeters, 2017-03-13:
Yeah, both @dirk-thomas and I can reproduce reliably with fresh docker containers, though with only version 0.9.1 of `ihmc_ros_core`

dljsjr, 2017-03-13:
@scpeters Does this build process use your gradle cache tarball?

scpeters, 2017-03-13:
Oh, I guess so. Let me try it without that tarball.

scpeters, 2017-03-13:
Yeah, it was that old gradle cache tarball. I'm just going to delete that. Sorry for the noise

dljsjr, 2017-03-14:
@scpeters No problem, just glad everything is working.

It's probably a good idea to make a fresh tarball every time we rev the IHMC Open Robotics Software binaries; you shouldn't need to modify the tarball in between changes to the ROS packages.