ihmcrobotics/ihmc-open-robotics-software #119: Should ModelFileLoader inherit ROS_PACKAGE_PATH?

dljsjr, 2017-02-28:
Because our software must be buildable both with and without ROS support, we have elected to take an approach where the core software is mostly indifferent of ROS and it interacts with ROS systems via various shims and runtime layers. Of note is that we don't get implicit access to most ROS-specific environment modifications, e.g. ROS_PACKAGE_PATH and the like, as well as not being privy to ROS-specific extensions to core functionality like custom URI schemes (e.g. package:// for indicating files on the ROS_PACKAGE_PATH, model:// for the GAZEBO_MODEL_PATH).

Of note, currently when we load model files (SDF, URDF), since we don't have implicit access to things like ROS_PACKAGE_PATH and GAZEBO_MODEL_PATH, we get around this by vendoring key assets in our source tree and then having a convention of "resource directories" attached to a model that identify sub-packages in the Java class path that contain relevant assets.

I am *not* proposing we get rid of this functionality, but I am wondering if maybe it would be useful to write the glue code that acquires things like ROS_PACKAGE_PATH from the process environment manually and maybe augments the resource directories with the environment path values.

Pros:
- Better integration with ROS

Cons:
- Our stuff still has to work without ROS so the code itself might get a little more complex since it's now handling both cases.
- We need to adopt either a convention or a workflow to establish what takes precedence when a resource is both on our class path and on the ROS_PACKAGE_PATH. It's pretty common to modify URDFs/SDFs with revised mass parameters or customizations, so whatever convention/workflow we pick we have to make sure that the precedence is obvious and well-documented.
- We'll have to refactor the whole resource location pipeline to differentiate between stuff on the class path and the filesystem. This should probably be done anyway and the whole thing should be refactored to use Java's built-in URL classes and URL Handlers (as awful as that API is) instead of us doing Stringly-typed hand parsing (that breaks all the time), but it's a lot of work

Thoughts? cc @jespersmith @calvertdw @btshrewsbury @SylvainBertrand

dljsjr, 2017-02-28:
Another Pro I forgot to mention: If we did this, it'd make it easier if we decided to try and support Xacro at some point in the future. But that's an entirely different discussion.

jespersmith, 2017-02-28:
We can have a complete ROS WS as submodule and use that for everything. That ros WS has to include the whole ROS world though.

dljsjr, 2017-02-28:
@jespersmith How do you mean?

jespersmith, 2017-02-28:
We build a ROS workspace with everything we depend on in it (so standard messages etc). We make that a git repo. We submodule that inside ModelFileLoaders and use that as our workspace path. Beats copying stuff into ModelFileLoaders and allows developers that have a Linux machine to checkout our "Canonical" ROS workspace and install new stuff in it.

dljsjr, 2017-02-28:
So we could use that to solve the vendoring/duplication problem and then switch everything over to only use ROS_PACKAGE_PATH?

jespersmith, 2017-02-28:
I would suppose so. It'll be a pretty big repo though. But not copied in our OSS library.

dljsjr, 2018-08-01:
@calvertdw Before the DDS rewrite I spent a bunch of time making the network configuration loadable from ENV and ROS_MASTER_URI would overwrite our config file's field. I don't know if that functionality was preserved when we switched to DDS but I agree that we should still be able to do that.

This ticket is specific to model loading though. This issue is trying to address the fact that our Robot Model definitions requires a `String[]` of "resource directories" to be able to find meshes and textures correctly. On a ROS 1/Gazebo system we could pull these from the ROS_PACKAGE_PATH/GAZEBO_MODEL_PATH ENV vars. But we'd have to solve the problem in a portable way to Windows/macOS for ROS 1 users and it might make the Robot Model code more complex instead of less.

closed 2023-12-30 by calvertdw