ihmcrobotics/ihmc-open-robotics-software #120: Slider doesn't update when COM height set via ROS API

wxmerkt, 2017-03-23:
| Required Info | |
|---|---|
|Version or commit hash|sylvain-maintenance-trip-local|
|Sim or Real Robot|Valkyrie Unit D|
|Operating System|Ubuntu 14.04|
|JDK|1.8.0_111|

#### Steps to reproduce issue
1. Set the COM height with slider 8
2. Update the COM height via the ROS API
3. Move slider 8 by a nudge

#### Expected behavior
The slider should be updated/moved when the ROS API modifies the corresponding value.

#### Actual behavior
Really rapid increase of COM as slider causes jump in value

#### Contributors with relevant knowledge or expertise (optional)
@dljsjr

dljsjr, 2017-03-23:
Are there any differences between `sylvain-maintenance-trip-local` and the `sylvain-maintenance-trip` branch that got merged in to the 0.9 release?

wxmerkt, 2017-03-23:
Here are the differences: https://github.com/openhumanoids/ihmc-open-robotics-software/compare/sylvain-maintenance-trip...openhumanoids:sylvain-maintenance-trip-local - nothing really that could affect the sliders

dljsjr, 2017-03-23:
Thanks, I wouldn't think there would be anything that would impact the sliders I just wanted a record of the diffs so we can figure out how to get you a patch or something

SylvainBertrand, 2017-03-23:
Yo! So it's my bad, the YoVariable that the slider board is hooked on is simply poorly updated in the controller. There is no straightforward fix at first glance as the height control is somewhat complicated (there are three types of input to deal with: from the walking controller, from the messages, and from the YoVariable). I'm afraid that a quick fix would result in introducing a new bug. We'll let you know if we can figure out a fix for it though. Sorry for the bad news.

wxmerkt, 2017-03-23:
Thanks for the explanation Sylvain - I see the issue. Is there a way to have a smooth transition instead if triggered by slider board? So if ``abs(current-desired)>epsilon`` it will have a maximum change rate?

wxmerkt, 2018-01-31:
Since 0.10, once a ROS message with the height has been set both the virtual and the actual sliderboard are not "responded" to anymore - i.e. once the height has been adjusted once via messages it cannot be adjusted via sliders anymore.

cc @VladimirIvan

dljsjr, 2018-01-31:
I believe at one point @georgwi was looking in to this

closed 2023-12-30 by calvertdw