ihmcrobotics/ihmc-open-robotics-software #121: Flaky startup of YoVariableProducer: "java.net.BindException: Address already in use"

scpeters, 2017-05-25:
| Required Info | |
|---|---|
|Version or commit hash| c787635debeb97ab02c44212954fd4aa9efb050b |
|Sim or Real Robot|Valkyrie sim in gazebo for SRC|
|Operating System|Ubuntu 14.04|
|JDK|openjdk version "1.8.0_111" (see below for more)|

~~~
$ java -version
openjdk version "1.8.0_111"
OpenJDK Runtime Environment (build 1.8.0_111-8u111-b14-3~14.04.1-b14)
OpenJDK 64-Bit Server VM (build 25.111-b14, mixed mode)
~~~

#### Steps to reproduce issue
Sometimes when starting the valkyrie sim, I see the following error in the console output and the controller doesn't seem to work. It was also reported on bitbucket:

* https://bitbucket.org/osrf/srcsim/issues/181/ihmc-robotdatalogger-javanetbindexception

~~~
Exception in thread "Thread-4" java.lang.RuntimeException: java.net.BindException: Address already in use
	at us.ihmc.robotDataLogger.YoVariableProducer.start(YoVariableProducer.java:148)
	at us.ihmc.robotDataLogger.YoVariableServer.start(YoVariableServer.java:117)
	at us.ihmc.avatar.networkProcessor.modules.ToolboxModule$2.run(ToolboxModule.java:149)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.net.BindException: Address already in use
	at sun.nio.ch.Net.bind0(Native Method)
	at sun.nio.ch.Net.bind(Net.java:433)
	at sun.nio.ch.Net.bind(Net.java:425)
	at sun.nio.ch.ServerSocketChannelImpl.bind(ServerSocketChannelImpl.java:223)
	at java.nio.channels.ServerSocketChannel.bind(ServerSocketChannel.java:157)
	at us.ihmc.multicastLogDataProtocol.MultiClientStreamingDataTCPServer.<init>(MultiClientStreamingDataTCPServer.java:43)
	at us.ihmc.multicastLogDataProtocol.MultiClientStreamingDataTCPServer.<init>(MultiClientStreamingDataTCPServer.java:55)
	at us.ihmc.robotDataLogger.YoVariableProducer.start(YoVariableProducer.java:143)
	... 3 more
~~~

#### Expected behavior
I don't expect to see that error and have the controller not work. Most of the time it doesn't happen.

#### Actual behavior
Sometimes it happens and the controller doesn't work. We have to restart the sim.

#### Contributors with relevant knowledge or expertise (optional)
@dljsjr

dljsjr, 2017-05-25:
@scpeters This is a socket problem. The YoVariableProducer is the component that streams data to the logger and remote visualizer, and if it can't bind then it can't start.

As the message implies, the address is already in use. Either something else on the system is using the address/port (probably not very likely), or a previously launched controller didn't shut down properly before the next attempt at launching.

jespersmith, 2017-07-14:
We're working on switching the logger data over to Fast-RTPS, which should sidestep this bug

dljsjr, 2018-08-01:
We have in fact switched to Fast-RTPS.

closed 2018-08-01 by dljsjr