ihmcrobotics/ihmc-open-robotics-software #122: When used for NASA Space Robotics Challenge, stability decreases as real time factor decreases

knoedler, 2017-06-04:
| Required Info | |
|---|---|
|Version or commit hash|c787635|
|Sim or Real Robot|R5 sim|
|Operating System| Ubuntu 14.04|
|JDK| openjdk version "1.8.0_111" |

#### Steps to reproduce issue
Steps to reproduce can be found at:
https://bitbucket.org/osrf/srcsim/issues/164/walking-controller-stability-decreases-as
A possibly related bug is at:
https://bitbucket.org/osrf/srcsim/issues/186/foot-glitches-near-table-on-sample-task-2
Please let me know if I can provide assistance on reproducing this.

#### Expected behavior
For the 164 case I would expect the controller to be able to walk in a stable manner if the real time factor is high or low.  For case 186 I would expect the controller to not glitch on that particular step.

Stable walking without regard to the real time factor of gazebo.

#### Actual behavior
Robot falls, knee wiggles or foot glitches depending on the real time factor of the simulator.

#### Contributors with relevant knowledge or expertise (optional)

dljsjr, 2017-06-05:
@knoedler The IHMC controller requires a "real time" environment, which in the context of simulation it means that the control threads need synchronization with the physics and sensor updates. So my guess would be that there's some sort of problem with the sync at low real time rates.

I know that OSRF and JSC are currently investigating some of the synchronization code for possible bugs, I'm not sure if this is related or not. We'll take a look as well.

cc @scpeters

dljsjr, 2017-06-21:
@scpeters Was this issue resolved when you synchronization code?

closed 2018-08-01 by dljsjr