ihmcrobotics/ihmc-open-robotics-software #123: When used for NASA Space Robotics Challenge, stability is impacted by pelvis yaw angle

knoedler, 2017-06-20:
| Required Info | |
|---|---|
|Version or commit hash|c787635|
|Sim or Real Robot|R5 sim|
|Operating System|Ubuntu 14.04|
|JDK|openjdk version "1.8.0_111"|

#### Steps to reproduce issue
Have the robot take steps to turn so the pelvis is at approximately PI radians yaw in world coordinates
Walk forward - Note the incorrect z axis corrections of the feet and likely falls

#### Expected behavior
The expected behavior would be that the yaw of the robot in world coordinates does not impact stability.

#### Actual behavior
The robot frequently mis-steps and falls when walking when the pelvis yaw angle is near PI radians.

#### Contributors with relevant knowledge or expertise (optional)
Please let me know if there is anything I can do to help you reproduce or collect data.

dljsjr, 2017-06-21:
This is almost assuredly an issue with the simulated IMU data coming in from the JSC API and/or Gazebo. The exact same state estimation algorithm runs on the real robot that runs in the SRC simulations and we have no issues with yaw in world coordinates as related to walking.

Data on the IMU from Gazebo would be tremendously helpful in confirming if I'm right though.

cc @scpeters

knoedler, 2017-06-21:
I assume you use the pelvis IMU data, so I have attached data from the middle and rear pelvis IMUs from Gazebo.  Let me know if there is other data that would be helpful.  The pelvis____Imu1.csv.gz data is during a turn from 0 to PI and the pelvis___Imu2.csv.gz data is walking with a yaw of PI.  Thank you for taking a look.

[pelvisMiddleImu1.csv.gz](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/1090171/pelvisMiddleImu1.csv.gz)
[pelvisMiddleImu2.csv.gz](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/1090172/pelvisMiddleImu2.csv.gz)
[pelvisRearImu1.csv.gz](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/1090173/pelvisRearImu1.csv.gz)
[pelvisRearImu2.csv.gz](https://github.com/ihmcrobotics/ihmc-open-robotics-software/files/1090174/pelvisRearImu2.csv.gz)

dljsjr, 2017-06-21:
Without even firing up a simulation, the orientation quaternion, when visualized, seems to be way off on the pelvis Rear IMU during the entirety of the travel. It doesn't seem to start at 0 nor does it seem to end at Pi for the yaw rotation.

Pelvis Middle IMU seems correct but Pelvis Rear IMU is what we use for state estimation (as well as one of the torso IMUs but we use that to filter spine joint velocities, not really for pose-in-world estimation).

I know that the pelvis rear IMU has a slight transform offset from the pelvis but it seems to be off-axis in all three dimensions which doesn't seem correct to me.

cc @scpeters once again, maybe he can give me some insight or do some visualizations.

dljsjr, 2017-06-21:
Whoops I lied. I mixed up XYZW and WXYZ when I was doing my math. The pelvis rear IMU seems to be good at the start and the end.

dljsjr, 2017-06-26:
So I can't see anything wrong with the data as directly exported from Gazebo. There's a chance that there's a problem with how the JSC API pulls the data out of Gazebo and packs it in to the ROS Control interface that provides it to our controller. We're gonna have to take some time to try and reproduce this ourselves.

knoedler, 2017-06-27:
Thanks for looking at the data.  Let me know if there is anything else I can do to help.

closed 2018-08-01 by dljsjr