ihmcrobotics/ihmc-open-robotics-software #124: Whole Body Trajectory Limited to 29 Points

knoedler, 2017-07-01:
| Required Info | |
|---|---|
|Version or commit hash|  c787635 |
|Sim or Real Robot| R5 sim - Gazebo |
|Operating System|Ubuntu 14.04|
|JDK|openjdk version "1.8.0_111"|

#### Steps to reproduce issue
Send a whole body trajectory of more than 29 points to the controller

#### Expected behavior
The trajectory is executed.

#### Actual behavior
The controller crashes.

#### Contributors with relevant knowledge or expertise (optional)
I have not personally reproduced this issue.  I am trying to submit the potential issues and features from the recently completed NASA Space Robotics Challenge.  The contact for this issue is Vinayak Jagtap - vvjagtap@wpi.edu

nmertins, 2017-07-03:
I've reached out to Vinayak asking for more info.

nmertins, 2017-07-24:
This problem has been addressed in our current develop branch. We are working on making a new release that will contain the fix.

closed 2017-07-24 by nmertins